Makuna/NeoPixelBus #302: compile error SPI.h: No such file or directory, - for ESP8266, Visual Micro.

perigalacticon, 2019-10-27:
**NOTE:  If you are seeking help or have questions, this is NOT the place to do it.**  For questions and support, jump on Gitter and ask away.  
[image]

**Describe the bug**
Just updated NeoPixelBus to version 2.5.3 and Arduino IDE to 1.8.10.
Now when compiling for Generic ESP8266 board using Visual Studio, I get the following errors:
However compiling using the Arduino IDE 1.8.10 it does not give an error.

`DotStarGenericMethod.h:119: In file included from
NeoPixelBus.h:69: from
Fire2012_FastLed_for_Minecraft_Skull_Poster_for_ESP8266.ino:16: from
 
Error compiling project sources
TwoWireSpiImple.h: 29:17: fatal error: SPI.h: No such file or directory
   #include <SPI.h>
   compilation terminated
Build failed for project 'Fire2012_FastLed_for_Minecraft_Skull_Poster_for_ESP8266'

**To Reproduce**
Steps to reproduce the behavior:
Compile sketch that was previously compiled with previous version of NeoPixelBus using Visual Studio.

**Expected behavior**
Compilation successful.

**Development environment (please complete the following information):**
 - OS:  Win10
 - Build Environment Visual Studio with Arduino IDE v.1.8.10].
 - Board target Generic ESP8266.
 - Library version 2.5.3

**Minimal Sketch that reproduced the problem:**

**Additional context**
I notice there are several SPI.h files in the Arduino libraries directory, one is for ESP8266, one is for ESP32, the other is in the unused libraries directory.

Makuna, 2019-10-28:
A very KEY point to your problem is that you are using Visual Micro; this is not something I specifically support.
Your sketch name is suspect, my library is NeoPixelBus, but that states FastLED???

Do you have more verbose output, like what SPI.h is being included? For Arduino IDE it outputs the following...
```
Using library SPI at version 1.0 in folder: C:\Users\<user>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SPI
```

Makuna, 2020-07-25:
closing due to no response and that the error is with FastLED not NeoPixelBus

closed 2020-07-25 by Makuna