Makuna/NeoPixelBus #303: With 2.5.1 NeoEsp8266BitBang800KbpsMethod stopped working

siedi, 2019-11-02:
Using the library for over three years now and wanted to upgrade my code to the latest version. Using NeoEsp8266BitBang800KbpsMethod with 228 WS2812b LEDs, with a custom made board and a stock ESP8266.
Worked up to version 2.5.0 of this library with no issues (even using bit-banging). With 2.5.1, the strip doesn't light up anymore. Going back to 2.5.0 works again. Even tried with a simple example, no Wifi or whatsoever.

**Expected behavior**
Should work with library version >2.5.0

**Development environment (please complete the following information):**
 - OS: Max
 - Build Environment Arduino 1.8.9
 - Board target ESP8266 Generic, v 2.5.2
 - Library version 2.5.0/2.5.1-2.5.3

**Minimal Sketch that reproduced the problem:**
```
#include <NeoPixelBus.h>
const uint16_t PixelCount = 228; // this example assumes 4 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 5;  // make sure to set this to the correct pin, ignored for Esp8266
#define colorSaturation 128
NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBang800KbpsMethod> strip(PixelCount, PixelPin);

RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

HslColor hslRed(red);
HslColor hslGreen(green);
HslColor hslBlue(blue);
HslColor hslWhite(white);
HslColor hslBlack(black);

void setup()
{
    Serial.begin(115200);
    while (!Serial); // wait for serial attach

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();

    Serial.println();
    Serial.println("Running...");
}

void loop()
{
    delay(5000);

    Serial.println("Colors R, G, B, W...");

    for (int i = 4; i < PixelCount; i++) {
      strip.SetPixelColor(i, blue);
    }
    strip.Show();

    delay(5000);

    Serial.println("Off ...");

    // turn off the pixels
    for (int i = 0; i < PixelCount; i++) {
      strip.SetPixelColor(i, black);
    }
    strip.Show();

    delay(5000);

    Serial.println("HSL Colors R, G, B, W...");

    // set the colors, 
    // if they don't match in order, you may need to use NeoGrbFeature feature
    strip.SetPixelColor(0, hslRed);
    strip.SetPixelColor(1, hslGreen);
    strip.SetPixelColor(2, hslBlue);
    strip.SetPixelColor(3, hslWhite);
    strip.Show();

    delay(5000);

    Serial.println("Off again...");

    // turn off the pixels
    strip.SetPixelColor(0, hslBlack);
    strip.SetPixelColor(1, hslBlack);
    strip.SetPixelColor(2, hslBlack);
    strip.SetPixelColor(3, hslBlack);
    strip.Show();

}
```

Makuna, 2019-11-02:
What do you mean by stopped working?  Please describe the problem.  No pixels light up?  Random effects?  Try to be specific

Do you have a means to capture the pin output like a Logic Analyzer or Oscilloscope? 

BTW, Bitbang is not a recommended method on the Esp, due to changes in the Esp core code around WiFi making it so it can not maintain a constant data stream for more a few pixels (some have had luck up to around 32 pixels).  I would highly recommend using one of the hardware methods, Dma or Uart; but they do have very specific pin usage requirements and its all covered in the Wiki.

Makuna, 2019-11-02:
https://github.com/Makuna/NeoPixelBus/pull/304

siedi, 2019-11-03:
Thanks so much for the quick fix. Indeed, works again, with the master branch / your PR included.

BTW, I'm running a wordclock with 228 LEDs with your library and bit-banging for over four years :-) (didn't know better when I built the custom PCB back then).

Issue can be closed.

siedi, 2019-11-03:
I'm sorry, still doesn't work as expected. Seems erratic. Most of the time, when I turn on the wordclock, no LED lights up, or maybe one or two (of the intended ones) with a random color. Even after 10 mins, nothing changes. 

When I turn on the party mode (all LEDs are turned on with a random, alternating color), even then, not all LEDs light up. But now comes the strange thing, after 20 seconds or so, out of sudden, it works as expected, so the colors change and all LEDs are on. 

When switching to the wordclock mode now, it works as expected.

Makuna, 2019-11-03:
Try using the `NeoEsp8266BitBangWs2812xMethod` method.

Makuna, 2019-12-09:
v2.5.4

closed 2019-12-09 by Makuna