Makuna/NeoPixelBus #305: Brightness not apply with NeoBitmapFile

jbreizh, 2019-11-07:
**Describe the bug**
Thanks for your work, but it seems thats Brightness is not apply with NeoBitmapFile

**To Reproduce**
take the NeoPixelBitmap exemple, but use a NeoPixelBrightnessBus object instead of a NeoPixelBus object and use SetBrightness() in the setup. When the animation run from the bitmap, the brightness is not set according to SetBrightness().

**Expected behavior**
SetBrightness() working in all situation

Makuna, 2019-11-07:
This is by design; on two different points.
1) BLT means Bit-aligned BLock Transfer, it works at low level to transfer bits, for speed.  
2) The NeoPixelbrightnessBus is used for legacy support for easier placement into sketches that were converted from other libraries.  It is not expected to be used with the more advanced features of this library as there are better ways to do this.
The only methods that support brightness modification are the `SetPixelColor` and `GetPixelColor` on the `NeoPixelBrightnessBus.`

**Solution:**
If you look at the NeoPixelBufferShader example, you will see an example on how to implement a brightness shader.  This is the prefered method to do this sort of thing.  
Then with the BitmapFile, just call the `Render` method instead of `Blt` using the above brightness shader.  
I would then suggest you also apply a gamma correction in the shader if you want better color matching of the bitmap.

closed 2019-11-07 by Makuna

jbreizh, 2019-11-08:
Thanks for the answer, i try to use it following the NeoPixelBufferShader example, but a little.. totaly lost.

First, my project (a light painting ESP8266 command from web resquest) work like the NeoPixelBitmap exemple.

so in my animation loop i change: `NEOBMPFILE.Blt(STRIP, 0, 0, ANIMATIONSTATE, NEOBMPFILE.Width());`
for : `NEOBMPFILE.Render<BrightnessShader<DotStarBgrFeature>>(STRIP, SHADER, 0, 0, ANIMATIONSTATE, NEOBMPFILE.Width());`
i think it's good, but the problem is in the Apply in BrightnessShader, i don't know what to do in it.
The NeoPixelBufferShader exemple is this (compile crash) :
`  void Apply(uint16_t index, uint8_t* pDest, uint8_t* pSrc)` 
 `  {` 
   ` // we don't care what the index is so we ignore it`
   ` //`
   ` // to apply our brightness shader, `
   ` // use the source color, modify, and apply to the destination `
    
   ` // for every byte in the pixel,`
   ` // scale the source value by the brightness and `
    `// store it in the destination byte`
    `const uint8_t* pSrcEnd = pSrc + T_COLOR_FEATURE::PixelSize;`
   ` while (pSrc != pSrcEnd)`
   ` {`
       ` *pDest++ = (*pSrc++ * (uint16_t(_brightness) + 1)) >> 8;`

he NeoPixelBufferShader exemple is this (compile work) :
`  RgbColor Apply(uint16_t index, RgbColor original)`
 ` {`
   ` // we don't care what the index is so we ignore it`
    `//`
   ` // to apply our brightness shader, modify the original color and return the color we want`
  `  // blend from black (_brightness == 0.0) to the original color (_brightness == 1.0)`
    
   ` return RgbColor::LinearBlend(Black, original, (float)_brightness / 255.0f);`
 ` }`

But it's way above my comprehension and my poor coding skill.... Can you help me my master?
I will send my code on github this evening, i just don't have time now.

Makuna, 2019-11-08:
I will take a glance at the code when you get a chance to send me a link.

Also, include the crash information (from the Serial output) and use the exception tool (ESP Exception Decoder, search and you will find it, it installs into Arduino IDE) to decode the crash information; this will provide the details of what crashed. 

I have a idea of what might be the problem, but get me the above and I can be more accurate.

jbreizh, 2019-11-09:
So i push my code on github : https://github.com/jbreizh/imagePaintingSTA
I look for ESP Exception Decoder, but it is use to track bugs but actually my code doesn't compile with Render instead of Blt.
This is my console return :
```
In file included from /home/jbreizh/Documents/Developpement/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:64:0,
                 from /home/jbreizh/Documents/Developpement/Projet/imagePainting/version2/ESP8266/imagePaintingSTA/imagePaintingSTA.ino:4:
/home/jbreizh/Documents/Developpement/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h: In instantiation of 'void NeoBitmapFile<T_COLOR_FEATURE, T_FILE_METHOD>::Render(NeoBufferContext<T_COLOR_FEATURE>, T_SHADER&, uint16_t, int16_t, int16_t, int16_t) [with T_SHADER = BrightnessShader<DotStarBgrFeature>; T_COLOR_FEATURE = DotStarBgrFeature; T_FILE_METHOD = fs::File; uint16_t = short unsigned int; int16_t = short int]':
/home/jbreizh/Documents/Developpement/Projet/imagePainting/version2/ESP8266/imagePaintingSTA/imagePaintingSTA.ino:433:117:   required from here
/home/jbreizh/Documents/Developpement/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:213:31: error: no matching function for call to 'BrightnessShader<DotStarBgrFeature>::Apply(uint16_t&, DotStar3Elements::ColorObject&)'
                         color = shader.Apply(indexPixel, color);
                               ^
/home/jbreizh/Documents/Developpement/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoBitmapFile.h:213:31: note: candidate is:
/home/jbreizh/Documents/Developpement/Projet/imagePainting/version2/ESP8266/imagePaintingSTA/imagePaintingSTA.ino:57:10: note: void BrightnessShader<T_COLOR_FEATURE>::Apply(uint16_t, uint8_t*, uint8_t*) [with T_COLOR_FEATURE = DotStarBgrFeature; uint16_t = short unsigned int; uint8_t = unsigned char]
     void Apply(uint16_t index, uint8_t* pDest, uint8_t* pSrc)
          ^
/home/jbreizh/Documents/Developpement/Projet/imagePainting/version2/ESP8266/imagePaintingSTA/imagePaintingSTA.ino:57:10: note:   candidate expects 3 arguments, 2 provided
exit status 1
Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini
```
I did commit with Blt (just change the line 432 with the 433). I think that the problem is in the line 57-72 (function Apply) but don't know what to change (i don't understand the code).
Thanks for your help, i am not far to complete my project.

Makuna, 2019-11-09:
https://github.com/jbreizh/imagePaintingSTA/pull/1
I fixed the code directly.  The specific shader needed had a different apply parameter list.  Similar to your thoughts above, you could use a linear blend to apply the brightness, but I included the fastest code.

Try this and see if it crashes.

jbreizh, 2019-11-09:
I merge your pull request and do some testing. It seems to work well, the animation take the brigthtness and the led are visibly dimm with the right color. So i call it a success.

clearTo doesn't take the brightness, but this is normal as it doesn't use the shader. I suppose, i have to make my own function for it. It's ok, i can do it if needed (it's a minor feature in my sketch).

Thanks you from France for your work and your reactivity.

jbreizh, 2019-11-10:
Hello, me again...
I try to use
`NeoPixelBus<DotStarBgrFeature, DotStarSpiMethod> STRIP(NUMPIXELS);`
instead of :
`NeoPixelBus<DotStarBgrFeature, DotStarMethod> STRIP(NUMPIXELS, CLOCK_PIN, DATA_PIN);`
but it don't seems to work in my sketch, but it work in DotStartTest exemple.
So i suspect a conflict with other librairy (especialy SPIFFS), can you confirme that i can't use DotStarSpiMethod with SPIFFS?

Makuna, 2019-11-10:
What do you mean it doesn't seem to work?  Crash?  Be specific.

Are you using the VERY latest Arduino/Esp82666 board support?  They just released a new version days ago.  There were some issues with SPIFFs that were fixed with it.

Are you correctly moving the Data and Clk wires over to the correct PINs, as SPI hardware only works on specific pins.

Try DotStarSpi2MhzMethod and see if it improves.

p.s.  Did you figure out that you can just call ClearTo like this to get shader support?
```
STRIP.ClearTo(  SHADER.Apply( 0, RgbColor(255, 255, 255) ) );
```

jbreizh, 2019-11-10:
Thanks for `STRIP.ClearTo(  SHADER.Apply( 0, RgbColor(255, 255, 255) ) );` . It work great.

For the DotStarSpiMethod, when i say "don't work" i mean output nothing, but no crash. The webserver still react and send confirmation, just no output.

I switch Data and Clock wires over D5(SCK) for Clock and D7(D7) for DATA. It seems to be good as i use a wemos mini (probably clone just label mini).

I don't use level shifter but it work in software....

I try again with the 2.6 version of ESP8266, but no change.

Anyway, will finish my "pretty" README.md and try to shoot some photo. I think that software Method will be good as i have only 60 led. Fastled wasn't making hardware and i use to stream every data through Serial and use WS2812/arduino/bluetooth. It turn out like that :
[image]
So i think it will be perfect.