Makuna/NeoPixelBus #306: NeoBitmapFile begin fail to open 2 times the same file

jbreizh, 2019-11-11:
**Describe the bug**
Hello, i think i have found a bug or a weird behavior. Follow this code (complete in zip attach).

```
void loop()
{ 
  loopCounter += 1;
  
  // string to hold result
  String msg = "Loop "+ String(loopCounter) + " : ";
  
  // Open the bitmap (good)
  if (loopCounter%2)  BMPFILE = SPIFFS.open("/pacman.bmp", "r");
  else BMPFILE = SPIFFS.open("/napoleon.bmp", "r");

   // Open the bitmap (error)
  //BMPFILE = SPIFFS.open("/pacman.bmp", "r");
  
  // Check and initialize bitmap from the file
  if (NEOBMPFILE.Begin(BMPFILE))
  {
    msg += "LOAD SUCCESS : Width=" + String(NEOBMPFILE.Width()) + "px Height=" + String(NEOBMPFILE.Height()) + "px";
  }
  else
  {
    msg += "LOAD FAILURE";
  }

  // Close the bitmap
  BMPFILE.close();

  //print msg
  Serial.println(msg);
  delay(500);
}
```
When i load one bitmap and then another :
```
if (loopCounter%2)  BMPFILE = SPIFFS.open("/pacman.bmp", "r");
  else BMPFILE = SPIFFS.open("/napoleon.bmp", "r");
```
It work :
```
Loop 670 : LOAD SUCCESS : Width=60px Height=154px
Loop 671 : LOAD SUCCESS : Width=60px Height=300px
Loop 672 : LOAD SUCCESS : Width=60px Height=154px
Loop 673 : LOAD SUCCESS : Width=60px Height=300px
Loop 674 : LOAD SUCCESS : Width=60px Height=154px
Loop 675 : LOAD SUCCESS : Width=60px Height=300px
Loop 676 : LOAD SUCCESS : Width=60px Height=154px
Loop 677 : LOAD SUCCESS : Width=60px Height=300px
Loop 678 : LOAD SUCCESS : Width=60px Height=154px
Loop 679 : LOAD SUCCESS : Width=60px Height=300px

```

But when i load the same bitmap :

` BMPFILE = SPIFFS.open("/pacman.bmp", "r");`

It fail 1 time on 2 time :

```
Loop 11 : LOAD SUCCESS : Width=60px Height=300px
Loop 12 : LOAD FAILURE
Loop 13 : LOAD SUCCESS : Width=60px Height=300px
Loop 14 : LOAD FAILURE
Loop 15 : LOAD SUCCESS : Width=60px Height=300px
Loop 16 : LOAD FAILURE
Loop 17 : LOAD SUCCESS : Width=60px Height=300px
Loop 18 : LOAD FAILURE
```

**Expected behavior**
To load the bitmap every time.

**Development environment (please complete the following information):**
 - OS: archlinux
 - Build Environment : Arduino IDE v.1.8.9
 - Board target : wemos D1 mini (core 2.6)
 - Library version : 2.5.3

**Minimal Sketch that reproduced the problem:**
in zip file
[test.zip](https://github.com/Makuna/NeoPixelBus/files/3832802/test.zip)

Thanks again for your work and have a good day.

Makuna, 2019-11-11:
I don't think this has anything to do with my library.

You are not checking the return value of the `SPIFFS.open` to make sure the file was open.  If it fails it will return null.
Also, turn on the debugging output; board settings, Debug Level, set to CORE I believe.  This will send debug output information.
Further, you are opening a file and not closing it.  You are using the same BMPFILE variable to open two different files, but you defined it not as a smart pointer, so it wont auto close on reassingment.  This makes no sense.

jbreizh, 2019-11-11:
i modify my loop like this :
```
void loop()
{ 
  loopCounter += 1;
  
  // string to hold result
  String msg = "Loop "+ String(loopCounter) + " : ";
  
  // Open the bitmap (good)
  //if (loopCounter%2)  BMPFILE = SPIFFS.open("/pacman.bmp", "r");
  //else BMPFILE = SPIFFS.open("/napoleon.bmp", "r");

   // Open the bitmap (error)
  BMPFILE = SPIFFS.open("/pacman.bmp", "r");

  
  if (!BMPFILE) {
    Serial.println("file open failed");
  }
  
  // Check and initialize bitmap from the file
  if (NEOBMPFILE.Begin(BMPFILE))
  {
    msg += "LOAD SUCCESS : Width=" + String(NEOBMPFILE.Width()) + "px Height=" + String(NEOBMPFILE.Height()) + "px";
  }
  else
  {
    msg += "LOAD FAILURE";
  }

  // Close the bitmap
  BMPFILE.close();

  //print msg
  Serial.println(msg);
  delay(500);
}
```
and no sign off opening failure :

```
Loop 243 : LOAD SUCCESS : Width=60px Height=300px
Loop 244 : LOAD FAILURE
Loop 245 : LOAD SUCCESS : Width=60px Height=300px
Loop 246 : LOAD FAILURE
Loop 247 : LOAD SUCCESS : Width=60px Height=300px
Loop 248 : LOAD FAILURE
Loop 249 : LOAD SUCCESS : Width=60px Height=300px
Loop 250 : LOAD FAILURE
Loop 251 : LOAD SUCCESS : Width=60px Height=300px
Loop 252 : LOAD FAILURE
Loop 253 : LOAD SUCCESS : Width=60px Height=300px
Loop 254 : LOAD FAILURE
Loop 255 : LOAD SUCCESS : Width=60px Height=300px
```
I don't understand when you say that i don't close the file, there is BMPFILE.close(); at the end of the loop. It doesn't close the file?

The smart pointer thing is way pass my knowledge (basic scripting), time for me to really learn C++

I will try for the debug output. Thanks

jbreizh, 2019-11-11:
Output debug level core :
```
LOAD SUCCESS : Width=60px Height=300px
SPIFFS_close: fd=1
SPIFFS_close: fd=1
SPIFFS_lseek rc=-10008
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 72 : LOAD FAILURE
SPIFFS_close: fd=1
Loop 73 : LOAD SUCCESS : Width=60px Height=300px
SPIFFS_close: fd=1
SPIFFS_close: fd=1
SPIFFS_lseek rc=-10008
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 74 : LOAD FAILURE
SPIFFS_close: fd=1
Loop 75 : LOAD SUCCESS : Width=60px Height=300px
SPIFFS_close: fd=1
SPIFFS_close: fd=1
SPIFFS_lseek rc=-10008
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 76 : LOAD FAILURE
SPIFFS_close: fd=1
Loop 77 : LOAD SUCCESS : Width=60px Height=300px
SPIFFS_close: fd=1
SPIFFS_close: fd=1
SPIFFS_lseek rc=-10008
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 78 : LOAD FAILURE
SPIFFS_close: fd=1
Loop 79 : LOAD SUCCESS : Width=60px Height=300px
SPIFFS_close: fd=1
SPIFFS_close: fd=1
SPIFFS_lseek rc=-10008
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 80 : LOAD FAILURE
SPIFFS_close: fd=1
Loop 81 : LOAD SUCCESS : Width=60px Height=300px
SPIFFS_close: fd=1
SPIFFS_close: fd=1
SPIFFS_lseek rc=-10008
SPIFFS_close: fd=1
SPIFFS_close: fd=1
```

jbreizh, 2019-11-11:
i try the same loop without :
```
   // Check and initialize bitmap from the file
  if (NEOBMPFILE.Begin(BMPFILE))
  {
    msg += "LOAD SUCCESS : Width=" + String(NEOBMPFILE.Width()) + "px Height=" + String(NEOBMPFILE.Height()) + "px";
  }
  else
  {
    msg += "LOAD FAILURE";
  }

```
and no more SPIFFS_lseek rc=-10008 in the debug :
```
Loop 88 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 89 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 90 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 91 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 92 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1

```
What does it mean... i don't know

Makuna, 2019-11-11:
if you search the source code of arduino/esp8266, the seek error is in the file spiffs_api.h line 426.  A seek request failed.
As part of the Bitmap loading (NeoBitmapFile.h, line 95), the file read position is set to the front the file (seek to zero position) and this is failing.  Trim the example down significantly (just simple file opens and seeks, no libraries other than SPIFF) to demonstrate the error and contact the esp8266/Arduino project.  
You may want to search if they have any active issues and get on their gitter channel and ask.

jbreizh, 2019-11-11:
you mean something like that :
```
#include <FS.h>
fs::File BMPFILE; 
int loopCounter = 0;

void setup()
{
  // FS setup
  SPIFFS.begin();

  // Serial setup
  Serial.begin(115200);
}

void loop()
{ 
  loopCounter += 1;
  
  // string to hold result
  String msg = "Loop "+ String(loopCounter) + " : ";
  
   // Open the bitmap (error)
  BMPFILE = SPIFFS.open("/pacman.bmp", "r");

  
  if (!BMPFILE) {
    Serial.println("file open failed");
  }

  BMPFILE.seek(0);
  BMPFILE.seek(100);
  BMPFILE.seek(0);
  
  // Close the bitmap
  BMPFILE.close();

  //print msg
  Serial.println(msg);
  delay(500);
}
```
Because it output no error :
```
SPIFFS_close: fd=1
Loop 1 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 2 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 3 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 4 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 5 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 6 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 7 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 8 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 9 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 10 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 11 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 12 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 13 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 14 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 15 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 16 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 17 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 18 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 19 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 20 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 21 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 22 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 23 : 
SPIFFS_close: fd=1
SPIFFS_close: fd=1
Loop 24 : 
```

Makuna, 2019-11-12:
but that isn't the same thing you are doing above.  Check closely, multiple file opens with the same handle.  You are not matching what you were doing.

jbreizh, 2019-11-12:
i am missing something....

I look the NeoBitmapFile.h and i see this :

```
  bool Begin(T_FILE_METHOD file)
    {
        if (_file)
        {
            _file.close();
        }

        if (!file || !file.seek(0))
        {
            goto error;
        }

        _file = file;
```

So if i understand Begin store its reference to file in _file and if _file already exist it close it before preceding with file.
So if my logic is good, it give for my loop :
```
loop 1 : _file not exist so nothing to close and _file = file
loop 2 : _file exist so it close it but as _file already equal to file it close file so problem
loop 3 : _file not exist so _file = file
```
....etc
so i modify my loop like this adding a  NEOBMPFILE.Begin(BMPFILE); after BMPFILE.close(); to be sure that file and _file are close :
```
void loop()
{ 
  loopCounter += 1;
  
  // string to hold result
  String msg = "Loop "+ String(loopCounter) + " : ";
  
  // Open the bitmap (good)
  //if (loopCounter%2)  BMPFILE = SPIFFS.open("/pacman.bmp", "r");
  //else BMPFILE = SPIFFS.open("/napoleon.bmp", "r");

   // Open the bitmap (error)
  BMPFILE = SPIFFS.open("/pacman.bmp", "r");
  
  // Check and initialize bitmap from the file
  if (NEOBMPFILE.Begin(BMPFILE))
  {
    msg += "LOAD SUCCESS : Width=" + String(NEOBMPFILE.Width()) + "px Height=" + String(NEOBMPFILE.Height()) + "px";
  }
  else
  {
    msg += "LOAD FAILURE";
  }

  // Close the bitmap
  BMPFILE.close();
  NEOBMPFILE.Begin(BMPFILE);
    
  //print msg
  Serial.println(msg);
  delay(500);
}
```

Is that correct? i don't feel pretty.... but it work.... i really need to change dynamically change the bitmap for my project.

jbreizh, 2019-11-17:
Just a message to say that i have solve all my problem...(i think).

For this problem of double bitmap, i had not understand that NeoBitmapFile keep track off the current file and close it when "he" begin with a new file. So no need to keep track of the file to close it. I was trying to do that to avoid multiple file open..

For my problem of hardware SPI, i just notice this evening that i forget to had "strip.Begin()" in the set-up. This was after testing every library i use to see if there was a cross interference and ordering a level schifter (3.3 to 5V) :-( :-( 3€ wasted lol . But why it was working in software without 
 strip.Begin() ? Do i need  Begin(int8_t sck, int8_t miso, int8_t mosi, int8_t ss) or Begin() is fine on ESP8266 ?

So i am happy, my project work great and is feature complete minus some code optimization especialy in the html+js interface. I will try to make an android app for it, it can be cool, i want to try this mobile web library like Onsen.

Take a look if you want https://github.com/jbreizh/imagePaintingSTA and thanks for your great library, it completely fits my needs.

closed 2019-11-27 by jbreizh