Makuna/NeoPixelBus #307: 'swap' is not a member of 'std'

nathanbv, 2019-11-25:
I used this library in the past without any issue. While trying to re-work a program after updating to th elatest version I found out that the code would not compile.
I figured that this issue came with the release 2.4.0 and the introduction of the support of the UART0.

When building using Arduino 1.8.9 on macOS 10.13.6 for a NodeMCU module, I get the following error on 2.4.0 and later releases:

```
In file included from /Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266UartMethod.cpp:28:0:
/Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266UartMethod.h: In member function 'void NeoEsp8266AsyncUart<T_UARTFEATURE, T_UARTCONTEXT>::UpdateUart()':
/Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266UartMethod.h:286:9: error: 'swap' is not a member of 'std'
         std::swap(_asyncPixels, _pixels);
         ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

2.3.5 builds fine and I don't mind using that release but I would rather find an explanation for this.
Could this issue come from an outdated std library on macOS ?

Makuna, 2019-11-26:
I believe the solution is to include stdlib before including my library; but I thought this was already done inside Arduino.h.  This specific issues keeps popping up due to the Arduino board work of reworking the namespacing.

I am on holiday and will slow to respond with an exact solution.

Makuna, 2019-11-27:
I can't repro your issue using the NeoPixelTest example and changing it to use the following definition so as to use uart.
```
NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart1Ws2812xMethod> strip(PixelCount, PixelPin);
```
Did you try to use the github version verses a released version of my library?

nathanbv, 2019-11-27:
Just by including the librabry the compilation error was shown.

However I found out the issue. By updating the Arduino IDE from 1.8.9 to 1.8.10, the error message got little more detailed (using the latest version 2.5.3 of NeoPixelBus):
```
In file included from /Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266UartMethod.cpp:28:0:
/Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266UartMethod.h: In member function 'void NeoEsp8266AsyncUart<T_UARTFEATURE, T_UARTCONTEXT>::UpdateUart(bool)':
/Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266UartMethod.h:272:9: error: 'swap' is not a member of 'std'
         std::swap(_pixelsSending, _pixels);
         ^
Multiple libraries were found for "NeoPixelBus.h"
 Used: /Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna
Multiple libraries were found for "SPI.h"
 Used: /Users/Nathan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SPI
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
Which is closer to the error from #243 thus when at start the IDE prompted me that a new version was available for the definition of the esp8266 board, I updated it from 2.3.0 to 2.6.1 (latest in date).
Rebuilding it was a success !

Funny thing is now the 2.3.5 version of the library does not build with the latest definition of the esp8266 board, throwing the following. But I get to use the latest revision of your library so that's perfectly fine.
```
In file included from /Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:67:0,
                 from /Users/Nathan/Documents/Arduino/testNeoPixelBus/testNeoPixelBus.ino:5:
/Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266DmaMethod.h:51:127: error: declaration of C function 'void rom_i2c_writeReg_Mask(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t)' conflicts with
     void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);
                                                                                                                               ^
In file included from /Users/Nathan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/esp8266_peri.h:25:0,
                 from /Users/Nathan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/Arduino.h:38,
                 from sketch/testNeoPixelBus.ino.cpp:1:
/Users/Nathan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/cores/esp8266/esp8266_undocumented.h:9:13: error: previous declaration 'void rom_i2c_writeReg_Mask(int, int, int, int, int, int)' here
 extern void rom_i2c_writeReg_Mask(int, int, int, int, int, int);
             ^
Multiple libraries were found for "NeoPixelBus.h"
 Used: /Users/Nathan/Documents/Arduino/libraries/NeoPixelBus_by_Makuna
Multiple libraries were found for "SPI.h"
 Used: /Users/Nathan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.1/libraries/SPI
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

For future reference, people who are having this issue, make sure you are up to date with the latest revision of the board definition (if you're not the Arduino IDE will pop a little window in the lower left corner when starting).

@Makuna thanks for your help and keep up the good work !
Hope this helps someone in the future.

closed 2019-11-27 by nathanbv