Makuna/NeoPixelBus #308: Change of behaviour in Class RgbColor methods .Darken() and .Brighten()

Rabter1, 2019-11-26:
When darkening a color with the method .Darken() the color is shifted/changed...

Issue:
Lets say i have a RgbColor(200,100,0) And i want to darken it lets say about half way...
To make things easy we call RgbColor.Darken(100)
Now.. what we get is a color with (100,0,0) so the color shifted into red...

I dont exactly know how RGB Colors work but it seems that you get away by reducing the same ammount of each color **relative** to each other.
So if i want to get half brightness of a color... i just take half of the each color-brightness
So out of RgBColor(200,100,0) i get (100,50,0)
For me at least it worked fine so far...

My suggestion for the method is:

-First check for the highest value in the color
-Then substract the ammount you want to reduce brightness
-Then devide though the highest value again so you get a multiplier
- Use this multiplier for the other colors...

Example: 
We have the color (250,100,25)
We reduce it by 100 --> color.Darken(100);

We do the math..
Highest value is 250
250-100 = 150
150/250 = 0,6
new color is 
250*0,6 = 150
100*0,6 = 60
25*0,6 = 15

I know its not perfect especially when yo get close to 0 and 255 this seems to induce also an error. but its way better then right now.. and a very fast fix. 
Maybe you like my way of Darkening and Brightening a color. 

Best regards Rabter

Makuna, 2019-11-27:
both the darken and lighten are legacy for compatibility with some old Adafruit sketches.
The best way to accomplish what you want is to keep the original source color (not previously modified) and to apply a blended version of it using a brightness level.  Sort of like this...
```
// produce a dimmed color of half the brightness of the original
RgbColor dimmed = RgbColor::LinearBlend(originalColor, RgbColor(0), 0.5f);
```
This way you change the dimming value and always get accurate value. 
If you tried to brighten a previously dimmed value, they will get a "muddied" color due to quantization of the numbers (think roundoffs).

Makuna, 2019-11-30:
To refrane from using floats (on AVR this can cause sketches in size greatly), I was thinking of adding this method, which you could use.  There will be no way to brighten, but as I previously commented, this isn't a good idea anyhow.

```
RgbColor RgbColor::Dim(uint8_t delta) const
{
    const uint8_t oldLevel = CalculateBrightness();
    const uint8_t newLevel = oldLevel - delta;

    return RgbColor(R * newLevel / oldLevel, G * newLevel / oldLevel, B * newLevel / oldLevel);
}
```

Makuna, 2019-12-04:
https://github.com/Makuna/NeoPixelBus/pull/312

Rabter1, 2019-12-05:
Thanks for the fast response and fix! 
Still.. i dont think it is very usefull to use Dim that way. 
Let me explain:
Lets say you want to Dim your Color but you dont know the brightness of it. So it is not usefull to Dim it by a fixed value because it could turn off the light even though you dont want to turn it off. Or you dont even see the difference... lets say 250-50 is still almost as bright as 200. 

I think it would be more usefull if you go by precentage...
So you Dim the light by 50% not by a fixed value.
Example
```
RgbColor RgbColor::Dim(float percentage) {

  percentage = max(percentage, 0); // Ensures it is at least 0
  percentage = min(percentag, 1); //Ensures it is not bigger than 1

    return RgbColor(R * percentage , G * percentage , B * percentage);
}
```

If you dont want to go by float and want to use uint8_t instead you could define 255 as actual brightness and 0 as turned off led 
Example:
```
RgbColor RgbColor::Dim(uint8_t AmmountOfDim) {

    return RgbColor(R * 255/ AmmountOfDim, G * 255/ AmmountOfDim, B * 255/ AmmountOfDim);
}
```

So if you have your color and you want to to have 20% brightness of it you could call
`newColor = oldColor.Dim(0.2);`
or
`newColor = oldColor.Dim(255*0.2);`

Or you go by uin8_t but are expecting values between 0 and 100 which would alse be very easy to use. But you lose a bit of resolution.

Makuna, 2019-12-05:
I guess you didn't look at the linked in pull that I merged in.  Because this is what it does; but the percentage is 0-255 where 255 is no change and 0 is total change.  Think of it is as "Dim To" rather than "Dim By".

this will dim the color "to" 50% as 127 is half of 255.
`newColor = oldColor.Dim(127);`

this will dim the color "by" 10% (to 90% calculated in place)
`newColor = oldColor.Dim(255 * 90 / 100);`
or if you want to continue to use floats
`newColor = oldColor.Dim(255 * 0.9);`

BTW, if you work with a small platform (like a AtTiny) you will want to avoid using floats like "0.2" because it natively doesn't support floats so it has to include a pretty large library of code to implement it, taking away valuable code space.  This is why the base features I try to stay away from it.

Rabter1, 2019-12-06:
Sorry to bother you. 
Now i see, that you did exactly that in your pull...
I didnt see because i used my phone. 
Well thanks!

Makuna, 2019-12-09:
v2.5.4

closed 2019-12-09 by Makuna