Makuna/NeoPixelBus #309: Nano 33 BLE (Sense) Support

odensc, 2019-12-01:
There are compilation errors with the Nano 33 BLE board:

```bash
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: error: 'F_CPU' was not declared in this scope
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:649:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT0h = ((uint32_t)(0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: note: suggested alternative: 'FPU'
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:649:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT0h = ((uint32_t)(0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: error: 'F_CPU' was not declared in this scope
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:650:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT1h = ((uint32_t)(0.80 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: note: suggested alternative: 'FPU'
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:650:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT1h = ((uint32_t)(0.80 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: error: 'F_CPU' was not declared in this scope
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:651:85: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t Cycles = ((uint32_t)(1.25 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                     ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: note: suggested alternative: 'FPU'
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:651:85: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t Cycles = ((uint32_t)(1.25 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                     ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: error: 'F_CPU' was not declared in this scope
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:675:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT0h = ((uint32_t)(0.50 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: note: suggested alternative: 'FPU'
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:675:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT0h = ((uint32_t)(0.50 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: error: 'F_CPU' was not declared in this scope
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:676:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT1h = ((uint32_t)(1.20 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: note: suggested alternative: 'FPU'
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:676:88: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t CyclesT1h = ((uint32_t)(1.20 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                        ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: error: 'F_CPU' was not declared in this scope
 #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:677:85: note: in expansion of macro 'ARM_OTHER_INST'
     static const uint32_t Cycles = ((uint32_t)(2.50 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                     ^~~~~~~~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:644:33: note: suggested alternative: 'FPU'
[... 111 lines cut ...]
         pmc_enable_periph_clk((uint32_t)TC3_IRQn);
                                                 ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:704:72: error: 'TC_Configure' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                                                                        ^
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:704:72: note: suggested alternative: 'nrf_pwm_configure'
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                                                                        ^
                                                                        nrf_pwm_configure
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:705:17: error: 'TC_Start' was not declared in this scope
         TC_Start(TC1, 0);
         ~~~~~~~~^~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:705:17: note: suggested alternative: 'va_start'
         TC_Start(TC1, 0);
         ~~~~~~~~^~~~~~~~
         va_start
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:753:16: error: 'TC_Stop' was not declared in this scope
         TC_Stop(TC1, 0);
         ~~~~~~~^~~~~~~~
/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoArmMethod.h:753:16: note: suggested alternative: 'i2c_stop'
         TC_Stop(TC1, 0);
         ~~~~~~~^~~~~~~~
         i2c_stop
exit status 1
```

Makuna, 2019-12-01:
Could you provide a few other lines from the compile above the errors?  I am specifically looking for a line that includes the defines, like a `-DARDUINO_ARCH_ARM` or similar.  It will be a long line with a bunch of other -D options.

Also, could you be specific about which device this is.  A link would help.

odensc, 2019-12-02:
Here are the defines:
```bash
-DARDUINO_ARCH_NRF52840 -DARDUINO=10809 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED
```

And the full line if you need it:
```bash
/home/oden/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -I. -g -Os -nostdlib @/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/defines.txt @/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -MMD -mcpu=cortex-m4 -DARDUINO=10809 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -I/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino -I/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE -I/home/oden/Arduino/libraries/ArduinoBLE/src -I/home/oden/Arduino/libraries/ArduinoJson/src -I/home/oden/Arduino/libraries/Arduino_LSM9DS1/src -I/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/libraries/Wire -I/home/oden/Arduino/libraries/NeoPixelBus_by_Makuna/src -I/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/libraries/SPI -I/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino/api/deprecated -iprefix/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/cores/arduino @/home/oden/.arduino15/packages/arduino/hardware/mbed/1.1.2/variants/ARDUINO_NANO33BLE/includes.txt /home/oden/Dev/Testing/bike/out/sketch/main.ino.cpp -o /home/oden/Dev/Testing/bike/out/sketch/main.ino.cpp.o
```

This is the exact device: https://store.arduino.cc/usa/nano-33-ble-sense

Thanks for looking into it!

Makuna, 2019-12-03:
I examined the code, and it would take significant work to support the NRF52* processors.  Since I don't have one this is not high on my list.

odensc, 2019-12-03:
Alright, thanks.

Makuna, 2019-12-04:
things I have learned:
1) bitbang won't work if BLE is used, since it would require the interrupts to turned off and the BLE won't work with them off.  Not turning them off will interrupt the bus signal timing.
2) PWM hardware feature is present that can be used and seems to the only way to accomplish this.

Makuna, 2019-12-06:
Heads up:  sometime tomorrow I will be merging in PWM driven support and I would appreciate any time you can take to try it and test it before I create an offical release next week.

odensc, 2019-12-06:
@Makuna Oh great - I'll be ready to test!

Makuna, 2019-12-06:
https://github.com/Makuna/NeoPixelBus/pull/313
merged

odensc, 2019-12-08:
@Makuna It's working fantastically over here! Nano 33 BLE Sense and WS2812B strip. Thanks for the great work!!

Makuna, 2019-12-09:
v2.5.4

closed 2019-12-09 by Makuna