Makuna/NeoPixelBus #310: Can i limited mA of power in neopixelbus?

tihoangyeudau, 2019-12-03:
Can i limited mA of power in neopixelbus? I dont wanna the power is overload

Makuna, 2019-12-03:
The mA draw is based on the number of pixels and colors you are showing.  If you have 100 pixels but they are showing black, it will draw none (or close to it).

Are you asking if there is a way to set a max mA and have the NeoPixelBus limit it?  Currently it does not.

tihoangyeudau, 2019-12-03:
Yes, i am asking a way to set a max mA to neopixelbus limit it. Have you anyway to do it?

Makuna, 2019-12-03:
Its an interesting concept, but I currently do not have such a feature.

I am unsure how you would even apply such a limit.  

Would you accept calls to "SetPixelColor" until it reaches the limit and then ignore all others?  Would you then ignore the very call "SetPixelColor" that would place the total usage over?
What happens when someone uses the "Pixels" method and writes directly into the buffer?  The same with using one of the DIBs/buffers which works in a similiar way?
Would you accept all pixel setting and instead when you call "Show" apply all pixels up to the limit and then restrict the other to black?
Since some methods only have one buffer, doing any of this would mean the colors retrieved by "GetPixelColor" may have set to black (off) to maintain the limit.

How are you thinking this would even work?

Makuna, 2019-12-04:
Maybe start out with a routine that calculates the mA of the given data?  This way the user can decide how they wish to react to adjust it.

The user would have to provide the approximate power usage of each led of a pixel, as they vary per manufacture and the Ws2811 are just controlling chips with no LEDs.  Further, there is 4 element leds like RGBW where the white can be drastricly different than the color LEDs.

```
// the passed in Color object will match the strip definition, RgbColor or RgbwColor and
// contian the milliamp rating for each LED for the matching color per spec from manufacturer.
// it will return the total milliamps that the strip will consume
uint32_t milliampsTotal = strip.CalculateMilliampsTotal( RgbColor(16, 16, 16)); 
// uint32_t milliampsTotal = strip.CalculateMilliampsTotal( RgbwColor(16, 16, 16, 20)); 
```

Makuna, 2020-01-12:
In the latest checkin, I included a routine to do the calculation.
```
uint32_t CalcTotalMilliAmpere(const typename T_COLOR_FEATURE::ColorObject::SettingsObject& settings)
```

and if you strip is RGB you would call it like...
```
// the leds take 16.5 milliamps peek, the values passed are in tenth of a milliamp
auto maTotal = strip.CalcTotalMilliAmpere( NeoRgbCurrentSettings(165,165,165) );
```

tihoangyeudau, 2020-01-14:
I will test it

closed 2020-01-21 by Makuna

antoniosap, 2023-08-20:
mmm

```
Compiling .pio/build/az-delivery-devkit-v4/src/main.cpp.o
src/main.cpp: In function 'void ledTestLights(RgbColor)':
src/main.cpp:132:79: error: no matching function for call to 'NeoPixelBus<NeoGrbFeature, NeoEsp32RmtMethodBase<NeoEsp32RmtSpeedWs2812x, NeoEsp32RmtChannel0> >::CalcTotalMilliAmpere(NeoRgbwCurrentSettings)'
   auto maTotal = strip.CalcTotalMilliAmpere(NeoRgbwCurrentSettings(1, 1, 1, 1));
                                                                               ^
In file included from src/main.cpp:7:
.pio/libdeps/az-delivery-devkit-v4/NeoPixelBus/src/NeoPixelBus.h:330:14: note: candidate: 'uint32_t NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::CalcTotalMilliAmpere(const typename T_COLOR_FEATURE::ColorObject::SettingsObject&) [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp32RmtMethodBase<NeoEsp32RmtSpeedWs2812x, NeoEsp32RmtChannel0>; uint32_t = unsigned int; typename T_COLOR_FEATURE::ColorObject::SettingsObject = NeoRgbCurrentSettings]'
     uint32_t CalcTotalMilliAmpere(const typename T_COLOR_FEATURE::ColorObject::SettingsObject& settings)
              ^~~~~~~~~~~~~~~~~~~~
.pio/libdeps/az-delivery-devkit-v4/NeoPixelBus/src/NeoPixelBus.h:330:14: note:   no known conversion for argument 1 from 'NeoRgbwCurrentSettings' to 'const SettingsObject&' {aka 'const NeoRgbCurrentSettings&'}
*** [.pio/build/az-delivery-devkit-v4/src/main.cpp.o] Error 1
```

Makuna, 2023-08-20:
@antoniosap It is inappropriate to comment on a closed issue like you have just done.  You should start with a QA discussion topic and reference this issue if you want; worst case is start an issue referencing this one. 
Make sure to include a snippit of your sketch along with platform and other details that are requested when creating an issue.