Makuna/NeoPixelBus #311: Parallel output for multiple platforms

Yordi-B, 2019-12-03:
Hi Guys,
Great job on the library, with all the added support.

Im currently using it for the RGBW color modes provided.

Since my LED project is getting bigger 2700 Leds. Im trying to find a solution to operate multiple parallel channels.

Ive noticed this is available for the ESP methods.

For the Teensy 3.2 im using FastLed provides a parallel output in combination with the octows2811 adapter.

Would it be possible to achieve this with neopixel, basicly suggesting an added feature for parallel output on teensy 3.X.

Makuna, 2019-12-06:
https://www.pjrc.com/teensy/td_libs_OctoWS2811.html

Yordi-B, 2019-12-10:
I'm unsure what you mean with the link.
Are you suggesting i use the OctoWS2811 library instead?

Good suggestion, unfortunately it doesnt support RGBWW I'm so desperately in need off.

Makuna, 2019-12-10:
Its just a reference link, nothing more.

Makuna, 2020-04-03:
already trackd by https://github.com/Makuna/NeoPixelBus/issues/160

closed 2020-04-03 by Makuna