Makuna/NeoPixelBus #317: NeoBitmapFile render take more time ... sometime

jbreizh, 2019-12-12:
Hello,
me again for my ligthpainting application... i make some picture to test my application. Nice picture i am pretty happy, but i notice some frame drop on some picture. So i try find the bottleneck and it seems to be NeoBitmapFile render that take 10+ time to complete than usual.
Here is my loop, but i mesure just the render time :
```
void updateAnimation(const AnimationParam& param)
{
  // Wait for this animation to complete,
  if (param.state == AnimationState_Completed)
  {
    // INDEX is in the limit
    if ((INDEXSTART <= INDEX) && (INDEX <= INDEXSTOP))
    {
      // Restart the animation
      ANIMATIONS.RestartAnimation(param.index);

      // Pause to do?
      if (ISPAUSE || ISCUT)
      {
        // Is it time to play?
        if ((PAUSECOUNTER > PAUSE))
        {
          // Initialisation
          PAUSECOUNTER -= 1;

          // Fil the strip : bitmap is crop to fit the strip !!!
          NEOBMPFILE.Render<BrightShader>(STRIP, SHADER, 0, 0, INDEX, NEOBMPFILE.Width());

          // Index
          if (ISINVERT) INDEX -= 1;
          else INDEX += 1;
        }
        // Is it time to wait?
        else
        {
          //Initialisation
          PAUSECOUNTER -= 1;

          // Blank or color the strip if needed
          if (ISENDOFF) STRIP.ClearTo(RgbColor(0, 0, 0));
          if (ISENDCOLOR) STRIP.ClearTo(SHADER.Apply(0, COLOR));

          // Cut the bitmap is needed
          if (ISCUT)
          {
            // Index
            if (ISINVERT) INDEX -= 1;
            else INDEX += 1;
          }

          // Have waited long enough?
          if (PAUSECOUNTER == 0) PAUSECOUNTER = 2 * PAUSE;
        }
      }
      // No pause to do !!! so let's play
      else
      {
        //
        ANIMETIME = millis();

        // Fil the strip : bitmap is crop to fit the strip !!!
        NEOBMPFILE.Render<BrightShader>(STRIP, SHADER, 0, 0, INDEX, NEOBMPFILE.Width());

        //
        Serial.println(millis() - ANIMETIME);

        // Index
        if (ISINVERT) INDEX -= 1;
        else INDEX += 1;
      }
    }
    // INDEX is out of the limit
    else
    {
      // Repeat to do?
      if (ISREPEAT && (REPEATCOUNTER > 0))
      {
        // Restart the animation
        ANIMATIONS.RestartAnimation(param.index);

        // Initialisation
        REPEATCOUNTER -= 1;

        // Index
        if (ISINVERT) INDEX = INDEXSTOP;
        else INDEX = INDEXSTART;
      }
      // Bounce to do?
      else if (ISBOUNCE && (REPEATCOUNTER > 0))
      {
        // Restart the animation
        ANIMATIONS.RestartAnimation(param.index);

        // Initialisation
        REPEATCOUNTER -= 1;
        ISINVERT = !ISINVERT; //invert the invert (following ??)

        // Index
        if (ISINVERT) INDEX = INDEXSTOP;
        else INDEX = INDEXSTART;
      }
      // Nothing more to do
      else
      {
        // Stop the animation
        ANIMATIONS.StopAnimation(param.index);

        // Blank or color the strip if needed
        if (ISENDOFF) STRIP.ClearTo(RgbColor(0, 0, 0));
        if (ISENDCOLOR) STRIP.ClearTo(SHADER.Apply(0, COLOR));
      }
    }
  }
```
for a bitmap of 119 pixel width (i miss my solder on one :-( ) it take mostly 2ms to complete but sometime it take 35ms complete so on the light painting picture it s very visible... Is there a rationnal explaination (buffer, seek) and a way to over past this issue.
thanks

jbreizh, 2019-12-12:
[image]
[image]

jbreizh, 2019-12-12:
More pixel, bigger bmp, more problem :-(
[image]

Makuna, 2019-12-12:
NeoBitmapFile reads directly from the file, so delays could be incurred from the file system (as you noticed, but the seeks and reads).

One way to confirm this is read the file into memory (call NeoBitmapFile::Blt to a instanced NeoDib) and use the NeoDib to render.  It will require more ram but should remove any latency from reading from the file.

jbreizh, 2019-12-12:
ok thanks, i just try to narrow the problem and it seems to be the 
` if (seek(xSrc, ySrc))`
so the seek that take mostly 0 ms and a few time 33ms to complete... weird but not related to your library. I will try your solution... why the life is so complicated

Makuna, 2019-12-12:
What file object are you using?  An SD card or something else?  Contact them to understand why the latency.

jbreizh, 2019-12-12:
the bitmap is on SPIFFS. I will search on the esp8266 arduino github if their is something related

Makuna, 2019-12-12:
This does seem strange for SPIFFS, as it just a memory pointer into the code memory area of flash.

jbreizh, 2019-12-12:
I change to use LittleFS, it seems to work without trouble. So SPIFFS problem, if it work with LitttleFS is good for me.
[image]

closed 2019-12-12 by jbreizh

Makuna, 2019-12-12:
Updated wiki to mentioned this and point back here
https://github.com/Makuna/NeoPixelBus/wiki/NeoBitmapFile-object-API

jbreizh, 2019-12-12:
ok thanks for your time.
I made some other picture, test and bug correction (mainly because SPIFFS start filename with a / and LittleFS don't). So LittleFS seems to be correct (i doesn't have seen any framedrop so far) and try some speed test by putting 0ms between each animation. So a bitmap of 119x360 take 300ms with LittleFS and nearly 1500ms with SPIFFS, crazy improvement !!! time to shoot picture outside, my appartement is too small for my new 2m ledbar.
[image]
[image]