Makuna/NeoPixelBus #319: ATtiny85 not compiling on version 2.5.2 and later

JasXSL, 2019-12-21:
I've used this library successfully on ATtiny85 before, but since 2.5.2 I'm getting the following compilation error:

`libraries\NeoPixelBus_by_Makuna\src/internal/DotStarGenericMethod.h:50:21: error: 'SCK' was not declared in this scope`

Win10, Arduino 1.8.10, ATtiny85 16mhz, v 2.5.2 and beyond.
Using the ATtiny board files located here: `https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json`

I'm assuming this is because there's no hardware SPI on it. Is there a workaround?

Makuna, 2019-12-21:
Two reasons, a missed spot that needed to be wrapped, and they are not defining __AVR_ATtiny85__ symbol in their build anymore.  
They use ARDUINO_attiny of which I have never seen before.  

New release coming shortly.

Makuna, 2019-12-21:
https://github.com/Makuna/NeoPixelBus/pull/320

closed 2019-12-21 by Makuna

hpsaturn, 2020-01-08:
This issue maybe also it is related with that in `PlatformIO` does not compile any sample with default PlatformIO config file for:

```python
[env:digispark-tiny]
platform = atmelavr
board = digispark-tiny
framework = arduino
lib_deps =
  547
```