Makuna/NeoPixelBus #322: Set brightness to 255 results in a wrap around

BlueAndi, 2019-12-25:
https://github.com/Makuna/NeoPixelBus/blob/e0782bb88b5937bf4d16c5004fdcfcf4c304da3b/src/NeoPixelBrightnessBus.h#L88

If ```brightness``` is 255, ```newBrightness``` gets 0 (wrap around).

https://github.com/Makuna/NeoPixelBus/blob/e0782bb88b5937bf4d16c5004fdcfcf4c304da3b/src/NeoPixelBrightnessBus.h#L102-L105

Scaling takes place with ```brightness```, not with ```newBrightness``` , which is ok.

https://github.com/Makuna/NeoPixelBus/blob/e0782bb88b5937bf4d16c5004fdcfcf4c304da3b/src/NeoPixelBrightnessBus.h#L121

But at the end the ```newBrightness``` is assigned to ```_brightness```, which is wrong.

---

https://github.com/Makuna/NeoPixelBus/blob/e0782bb88b5937bf4d16c5004fdcfcf4c304da3b/src/NeoPixelBrightnessBus.h#L95

Will result in a wrap around in case the ```_brightness``` is 0.

Makuna, 2019-12-26:
What actual problem are you seeing?

The wrap around is intentional and works both ways, check the method to get the brightness, it will take the stored value and subtract one, so the zero gets returned as 255 (wrapping the other way).

BlueAndi, 2019-12-27:
Yes, my fault. I missed the GetBrightness().
https://github.com/Makuna/NeoPixelBus/blob/e0782bb88b5937bf4d16c5004fdcfcf4c304da3b/src/NeoPixelBrightnessBus.h#L126-L129

closed 2019-12-27 by BlueAndi