Makuna/NeoPixelBus #324: WS2801 support

Aircoookie, 2019-12-28:
Hi @Makuna,
I hope you are having a very merry Christmas!
Surprised that no one suggested this before, but is it possible to drive WS2801?
They are one of the oldest addressable strip types and have been largely superseded by WS2812s, but there are still a lot of them around and quite a few of my (potential) users would like them supported: https://github.com/Aircoookie/WLED/issues/178
They are 2 wire and are likely using a similar SPI-like protocol as APA102.

There is already a library to facilitate driving them:
https://github.com/adafruit/Adafruit-WS2801-Library
I would consider using it, but thought that support for this driver IC would also be a great enhancement to your library. My apologies if you are already aware of it :)

Thank you and have a good start into the new decade!

Christian

Makuna, 2019-12-29:
I don't have one of these but it should be doable; would you be willing to test it?  What platform do you use?

Aircoookie, 2019-12-30:
Thank you for the quick response!
Yes, definitely willing to test and give feedback! Don't own these yet though so it will take a few days for them to arrive. My WLED sketch runs on ESP8266 primarily, though ESP32 is also supported.

Makuna, 2019-12-31:
https://github.com/Makuna/NeoPixelBus/pull/326
// for software bit bang
NeoPixelBus<NeoRgbFeature, NeoWs2801Method> strip(PixelCount, DotClockPin, DotDataPin);

// for hardware SPI (best performance but must use hardware pins)
//NeoPixelBus<NeoRgbFeature, NeoWs2801SpiMethod> strip(PixelCount);

Aircoookie, 2019-12-31:
Amazing, thanks!
Will test both methods once the hardware arrives :)

insensitiveclod, 2020-01-06:
Zomg. I just discovered WLED via Hassio/Homeassistant ,looking for ways to add led-strips; saw the 'QuinLed' boards ... and...
I still have about 50 meters of 2801 strip laying about, so I was wondering how to adapt them to 2801 function; thinking of just how much dev-work that'd also require.

And then, 2 days before I imagine'd I'd need this, the dev of WLED and of NeoPixelBus get together in a mighty duo and do-that-work.
Awesome!
I'll report back my findings when I get time/opportunity to hook up a few ESP's to my strips.
Cheers!

Aircoookie, 2020-01-10:
My WS2801 finally arrived today!
@Makuna the support works very well, thank you!
No flickering or major impact on WiFi performance.
Tested bitbang and hardware SPI, both are working fine!
Only have 32 LEDs here, but they worked as expected even when setting the led count up to 1500! (of course that badly hurts the framerate, these work best with up to 250 LEDs), so everything is nice.
This can be closed. It would be nice if you could make a minor release so users could update via library manager, but master works as well for now :)

Thanks once again! I really appreciate your work! (also donated a beer or two after Christmas for the cause)

@insensitiveclod Expect support in WLED master very very soon :)

Makuna, 2020-01-10:
v2.5.6 release just made

closed 2020-01-10 by Makuna