Makuna/NeoPixelBus #325: TM1814 RGBW Support

theomegapixel, 2019-12-30:
Hello Makuna,
I was wondering if it were possible to add support for the TM1814 IC RGBW [https://datasheetspdf.com/pdf-file/1330606/TitanMicro/TM1814/1](url) on the ESP32. I've previously been using the adafruit neopixel library and a teensy3.5 and modified the library to match the data sheet it seems fairly simple but I'm unfamiliar with your library. 

I used the SK6812 RGBW constructor and added the required C1 and C2 values to the frame at the begging of the buffer and adjusted the timings to match, I ended up with `pixels.setPixelColor(amp, ~amp, 3 pixels.color(255,0,0,0); and pixels.fill(amp, ~amp, pixels.color(255,0,0,0), 0, NUM_LEDS);`. I then use a NOR gate to invert the signal.

Makuna, 2019-12-31:
Your link didn't work.  [This link to  TM1804 spec sheet](https://datasheet.lcsc.com/szlcsc/1810010413_TM-Shenzhen-Titan-Micro-Elec-TM1804_C88266.pdf) does NOT describe anything outside of a normal signal, so the Neo800MhzMethod should work.

Why are you inverting the signal?

Makuna, 2019-12-31:
NOTES:  These chips are unique in the ability to set the CC for each channel of the LEDs (but shared with all on the strip).  These value are sent at the front of the data stream.  There will be a need to expose setting these value.

theomegapixel, 2020-01-01:
Hey Michael,

I hope you had a good New Years!
Thanks for taking a look at this, if you do implement these chips I can report back with tests and oscilloscope readings.

Thanks,
Connor

theomegapixel, 2020-01-04:
hey Michael,

the link you sent is for the 1804 NOT the 1814 chip set, I've attached a
copy of the 1814.
The 1814 has a resting high state for the reset so I use the NOR gate
rather than modifying the adafruit neopixel library further.

ryanjjennings, 2020-01-06:
Makuna,

Thanks for taking a look at this.  I have some I would like to run in Wled.

Ryan

Makuna, 2020-01-09:
I think I have a solution that doesn't require a bunch of code duplication; but it will take a bit to spread it through all the methods for all platforms.  initial work shows a minor code increase for other methods/features that don't have these "settings" but no extra memory for them.

theomegapixel, 2020-01-11:
Thanks Michael for giving this a shot I'm looking forward to testing the code.

Makuna, 2020-01-12:
https://github.com/Makuna/NeoPixelBus/pull/329
Esp32 Support is now checked in.  I will continue to work on other platforms before creating a new release.  Please try this out.
The normal way to define the NeoPixelBus to use Tm1814 is like the following (note, no invert hardware required, the normal signal is already inverted as required for this chip.
```
NeoPixelBus<NeoWrgbTm1814Feature, NeoTm1814Method> strip(PixelCount, PixelPin);
```
if you are using hardware to invert the signal, then you can use the following definition which will invert the signal from the normal required TM1814 signal.
```
NeoPixelBus<NeoWrgbTm1814Feature, NeoTm1814InvertedMethod> strip(PixelCount, PixelPin);
```

To set the constant current feature these chips support, use the following line once somewhere in setup or loop
```
strip.SetPixelSettings(NeoTm1814Settings(165,165,165,380)); // RGBW currents, in tenth milliamperes
```
Of course, change the numbers to the current requirements of your LEDs.

Makuna, 2020-01-12:
I am also interested if using the settings could be used to dim all the LEDs,  running them at half their rated current would mean 50% dimmer and might increase the effective range of the RGB values; where normally its hard to tell 200,200,200 from 255,255,255.

theomegapixel, 2020-01-14:
Hi Michael,

thank you very much for adding the support. So far the tests work well there aren't any major problems. So far so good, I'll continue to test and see if any issues arise.

I've tried using the current controls it can be subtle the difference in mA output but yes you can you it to extend the range. I'm yet to do test with an accurate colour meter or spectrometer to see what the affects are.

Thanks  again,
Connor

closed 2020-01-21 by Makuna