Makuna/NeoPixelBus #327: Board Support for FireBeetle ESP32

rclay701, 2020-01-04:
I am attempting to use the NeoPixelBus library and the FireBeetle ESP32 board but I am seeing the "Platform Currently Not Supported, please add an Issue at Github/Makuna/NeoPixelBus" message.  Are there any plans to support this board? 

My code includes:
#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

I have loaded the FireBeetle board and your library but the code will not compile without a definition for the FireBeetle board.  I can change to an Arduino Uno that I have and it compiles and loads correctly.  I need to use the FireBeetle for power consumption reasons.

Thanks

Makuna, 2020-01-05:
Did you try the a generic Esp32 Dev Board?
When it compiles, the verbose output will include a long line that contains values passed to the compilers, one should be "-DARDUINO_ARCH_ESP32".  If you don't find this (and based on your comments above you will not), then copy the complete line and post it here.  
It should have "ARDUINO_ARCH_ESP32" defined if its an ESP32 board supported for Arduino.

rclay701, 2020-01-05:
Michael - Thank you for the reply.  I took your advice and loaded the
boards in the Arduino-ESP32 json file.  I selected a few different boards
and the code compiled and loaded successfully.  I then selected the
FireBeetle board and it also worked.  There must be board definitions in
both json files (my original board definition came directly from the
DFRobot website).  I will verify it is in both jsons.  I changed the link
in the Preferences, additional board managers back to the DFRobot link and
removed their board manager and reloaded the managers from
the Auduino-ESP32 package on GitHub and the FireBeetle board was there.  It
still compiles and loads successfully.  I am perplexed but moving forward.

Rick

rclay701, 2020-01-05:
Michael - One additional tidbit....the GitHub stub was under expressif.

RIck

rclay701, 2020-01-05:
I confirmed the json from DFRobots does not work with my code.  The expressif board library does work.

closed 2020-01-05 by rclay701