Makuna/NeoPixelBus #333: Calling StopAnimation()/RestartAnimation() From Interrupt Causes Unexpected Results

mriksman, 2020-01-20:
**Describe the bug**
My original code used a web server to change animations. A button would stop all animations, and then start the next one. This worked well.

I then added an interrupt handler to GPIO0 and called the same function to change animations. The results are unexpected. The DEBUG says .IsAnimationActive is FALSE, but the animation is clearly still running.

There are 4 animations; 3 are unique (the 4th runs alongside another). 
  0
  1
  2 (which automatically starts 3; they work together).

**To Reproduce**
Run my code. (Set SSID/Password).
Open the page esp32.local. It will start the next animation. Refresh the page; the next animation starts. Cycle through a few times - it works well.
Now press the GPIO0 button. Watch the logs and the animations. Things are messed up. Animations don't stop, animations freeze. 

**Expected behavior**
GPIO0/interrupt calls to the Animation API should behave the same way as normal code.

**Development environment (please complete the following information):**
 - OS: Win10
 - Build Environment Arduino IDE v.1.8.10
 - Board target ESP32 Dev Module
 - Library version 2.5.3

**Minimal Sketch that reproduced the problem:**
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

WebServer server(80);

const char* ssid     = "3020";
const char* password = "1234567890";

const int reset_gpio = 0;

#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

//Number of animations 
#define ANIMATION_COUNT   4

enum animationNames {
  rainbowIdx      = 0,
  smoothBlipIdx   = 1,
  moveIdx         = 2,
  fadeIdx         = 3
};

const uint8_t PixelPin = 13; 
//Set stair width and number of steps
const uint8_t StepWidth = 60;  
const uint8_t NumSteps = 5;
const uint16_t PixelCount = StepWidth * NumSteps;

NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
NeoPixelAnimator animations(ANIMATION_COUNT); 

typedef RowMajorLayout MyPanelLayout;
NeoTopology<MyPanelLayout> topo(StepWidth, NumSteps);

struct AnimationController {
  float dimmer;  // controls overall brightness 0-1f
  float offset;  // relative offset of this blip
  float width; // width of the blip
  float hue[10]; // color of the blip
  int8_t  direction;
  uint16_t nextAnimation;
};
[... 286 lines cut ...]
    // Set GPIO 0 up as a way to reset Homekit configuration
    pinMode(reset_gpio,INPUT_PULLUP);
    attachInterrupt(digitalPinToInterrupt(reset_gpio), reset_intr_callback, RISING);

    strip.Begin();   
    strip.Show();
  
    animationsInit();

    if (MDNS.begin("esp32")) {
      Serial.println("MDNS responder started");
    }
    server.on("/", handleRoot);
  
    Serial.println("HTTP server started");
    server.begin();

}

void loop() {
  animations.UpdateAnimations();
  strip.Show();

  server.handleClient();
}

Makuna, 2020-01-20:
You are assuming that the animations are thread safe and they are not.  On the Esp32, IRQs are not always called back on the same core as the setup/loop are run on.  The variables for the state of the animation are not maintained by the compiled code to be consistent across the cores.
Further, its not safe to call MANY list based objects (lists of animations in this case) from IRQs without putting in place semaphore/lock system to protect them.  This is left up to the caller to do this, as this is a very rare need on Arduino and many Arduino platforms do not provide a way to do it.

Thus, the simplest and most common way is to set a flag in the IRQ and check this from loop to apply the action.

mriksman, 2020-01-20:
You are a certifiable genius. I used a `bool `to flag a required animation change and let the `loop()` do the work. Works like a charm.
I barely understood what you are saying... 'Thread safe'? How does one make code 'thread safe'?
The reason I picked this issue up is I am using the `ESP-HomeKit` code. So it too must be calling from a different core/thread/interrupt/something?

closed 2020-01-20 by mriksman

Makuna, 2020-01-20:
[topic on multithreading ](https://en.wikipedia.org/wiki/Multithreading_(computer_architecture))
[topic on multi-core](https://en.wikipedia.org/wiki/Multi-core_processor)

The Esp32 is pretty unique for Arduino as it is one of the first multi-core; while multi-threading has been around in a few other Arduino hardware platforms it isn't critical for them, since the Esp32 has wifi that primarily functions on the second core.