Makuna/NeoPixelBus #335: NeoPixelBus should with GPIO2 on my nodemcu V3 (esp8266)

stoefln, 2020-01-21:
**Is your feature request related to a problem? Please describe.**
I used fastLed before and used GPIO 4 to control my strip. Assuming that this would work also with other libraries I built a couple of LED control boards.
Now I would like to switch to NeoPixelBus, but the documentation is telling me that only GPIO1 - 3 are supported.

**Describe the solution you'd like**
Would be great to support GPIO4

**Describe alternatives you've considered**
A. Rewiring my hardware, but that's not really an option (I have 25 of them)
B. Using fastLed. I really don't want to do this either, cause I have a big codebase built on top of NeoPixelBus which I would like to use for my project. Also I like favour API of NeoPixelBus quite a bit.

Makuna, 2020-01-22:
Unfortunately, there is no reliable solution other than the hardware methods, and they have very restrictive pin support. What FastLED does is not a safe or reliable method on esp8266.

You can try the `NeoEsp8266BitBangWs2812xMethod` which will allow you to use any pin.  BUT this is not very reliable due to WiFi interrupts and there is no way to safely lock the interrupts out as it will cause WiFi to fail (from crashes to general connection failures).  I believe this is similiar to FastLed implementation.

penfold42, 2020-01-22:
I suggest you go for : 
A. Rewiring my hardware, but that's not really an option 

If you make GPIO4 an input you only need 1 bodge wire from GPIO4 to one of the supported pins.

stoefln, 2020-01-22:
> You can try the `NeoEsp8266BitBangWs2812xMethod` which will allow you to use any pin. 
Thanks! I would like to try...
Could you tell me how to initialize the strip? How can I set the data pin?
What I currently have is:
`NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart0800KbpsMethod> strip(pixelCount);`

stoefln, 2020-01-22:
> Could you tell me how to initialize the strip? How can I set the data pin?

Figured it out:
> `NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart0800KbpsMethod> strip(pixelCount, gpio);`

So what's very confusing for me now is this:
With fastled I use value "4" as pin and aparently I need to use value "2" with the makuna pixel bus. "4" does not work.
That's counter intuitive considering this table:
```
/*
  NodeMcu IO Index          = Esp8266 GPIO
  static const uint8_t D0   = 16;
  static const uint8_t D1   = 5;
  static const uint8_t D2   = 4;
  static const uint8_t D3   = 0;
  static const uint8_t D4   = 2;
  static const uint8_t D5   = 14;
  static const uint8_t D6   = 12;
  static const uint8_t D7   = 13;
  static const uint8_t D8   = 15;
  static const uint8_t D9   = 3;
  static const uint8_t D10  = 1;

*/
```
Can anybody enlighten me please?

Makuna, 2020-01-22:
Again, on Esp8266, the Dma & Uart methods have a hardcoded pin, although you can pass what ever you want it will ignore it.  In your last comment you are asking about a pin but your example is using the uart method.

The pin passed into the strip constructor is the GPIO pin.  Blame all the esp8266 board makers for the confusion of using D2 label on their boards.  

The Esp8266 GPIO4 (or just 4 when passed to the NeoPixelBus constructor) is the same as D2 (as labeled on the board); as the board file you included assigns.  So you can use `D2` or just `4` for the GPIO4 pin for your board.

stoefln, 2020-01-24:
Damn, made a copy-paste-mistake in my comment above. What I figured out was of cause:
> NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBangWs2812xMethod > strip(pixelCount, gpio);

> The Esp8266 GPIO4 (or just 4 when passed to the NeoPixelBus constructor) is the same as D2 (as labeled on the board); as the board file you included assigns. So you can use D2 or just 4 for the GPIO4 pin for your board.

I know all of that. I am just confused cause I am using value "4" for fastled and value "2" for pixelbus.
My problem is resolved now, thanks a lot for your help!

Makuna, 2020-01-24:
What? I just checked two of my boards, passing 4 will effect the GPIO 4 pin.  

NeoPixelBus does no translation of that number, it passes it through to the system.

closed 2020-01-24 by Makuna