Makuna/NeoPixelBus #339: NeoPixelBrightnessBus setBrightness overrides APA102 global brightness byte

feindsender, 2020-02-24:
Hi,

when using the setBrightness function with APA102 strips the global brightness byte of the strips
get changed. Also this changed byte get sent through SPI on strip.Show. This leads to flickering.
I tried to debug this at:
https://wled.discourse.group/t/leds-flickering-while-using-brightness

Also it would be nice if you could support the little changes for the SK9822 strip detailed here:
https://cpldcpu.wordpress.com/2016/12/13/sk9822-a-clone-of-the-apa102/

greetings
Maik

Def3nder, 2020-02-24:
Hi @Makuna, 
if I did understand @feindsender right as I discussed this in the mentioned WLED discord, he's using a LBGR strip.

feindsender, 2020-02-24:
Hi,
no i am using just a simple RGB in this case BGR strip.
Also i do not think there are any RGBW strips with APA102 leds.
Google only shows strips with alternating RGB and WHITE leds.

Makuna, 2020-02-24:
The DotStar variants include a per pixel modification bits along with the color data; and yes, the NeoPixelBrightnessBus does not honor those bits.

Makuna, 2020-02-25:
https://github.com/Makuna/NeoPixelBus/pull/340
This should fix it.  Give it a try and let me know.

feindsender, 2020-02-25:
Hi,
thanks for trying to fix it.
I am using the NeoPixelBrightness example sketch modified for the Dotstar LED.
With it i cannot get the LEDs to light at all.
```
#include <NeoPixelBrightnessBus.h> // instead of NeoPixelBus.h

const uint16_t PixelCount = 4; // this example assumes 3 pixels, making it smaller will cause a failure
const uint8_t DotClockPin = 14;
const uint8_t DotDataPin = 13;  

#define colorSaturation 255 // saturation of color constants

NeoPixelBrightnessBus<DotStarBgrFeature, DotStarMethod> strip(PixelCount);
RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation, colorSaturation, colorSaturation);

// you loose the original color the lower the dim value used
// here due to quantization
const uint8_t c_MinBrightness = 8; 
const uint8_t c_MaxBrightness = 254;

int8_t direction; // current direction of dimming

void setup()
{
    Serial.begin(115200);
    while (!Serial); // wait for serial attach

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();

    direction = -1; // default to dim first
    
    Serial.println();
    Serial.println("Running...");

    // set our three original colors
    strip.SetPixelColor(0, red);
    strip.SetPixelColor(1, green);
    strip.SetPixelColor(2, blue);
    strip.SetPixelColor(3, white);
    
    strip.Show();
}

void loop()
{
    uint8_t brightness = strip.GetBrightness();
    Serial.println(brightness);
    
    delay(500);

    // swap diection of dim when limits are reached
    //
    if (direction < 0 && brightness <= c_MinBrightness)
    {
      direction = 1;
    }
    else if (direction > 0 && brightness >= c_MaxBrightness)
    {
      direction = -1;
    }
    // apply dimming
    brightness += direction;
    strip.SetBrightness(brightness);

    // show the results
    strip.Show();
}
```
Using WLED the strips lights up but using the brightness control gives strange results.
Setting Brightness to 255 turns off the Strip.
Turning the Strip off and it glows at the lowest level.
Dimming up the LEDs flicker.
Dimming down works fine.

Makuna, 2020-02-25:
I will check this out.

BTW.  I really don't expect anyone other than novices to use NeoPixelBrightnessBus.  It was only meant for compatibility with Adafruits NeoPixel.  The math around the abstractions causes issues that not obvious (quantization) which manifest when using extreme cases without a way to handle them.  What I recommend is to use a "buffer" to draw into (calling its SetPixelColor), then Render it using a shader to the bus.  You will get much better results and consistent dim and brighten results; but it will cost you memory to implement the buffer.  There is an example for this.

Aircoookie, 2020-02-27:
@Makuna thank you for looking into it!

Of course I am aware that the `NeoPixelBrightnessBus` does has its limitations. Quantization does occur, but at least in my specific use case it has not been a major issue (I'm not generally doing brightness changes without setting the correct colors again afterwards). Using a buffer would of cause be the optimal solution to get accurate results from GetPixelColor(), but memory-wise having another buffer is not justifiable (at least on a complex ESP8266 application with 15k free heap that aims to control 1000+ LEDs, on a smaller scale project it would of course work). So I know that you recommend against using the BrightnessBus, but if it wasn't offered, I'd rather have to scale the channels down myself and just refrain from using any GetPixelColor(). I have to admit that WLED is a very nieche and unique use case (probably also one of the only pieces of code that combine using both NeoPixelBus for its amazing stability, RGBW support and large amount of supported LED types as well as FastLED for the color palettes and easy color magic)

Thank you once again for this amazing library, without it a lot of pixel projects would be non-existent or a lot worse!

Makuna, 2020-02-27:
https://github.com/Makuna/NeoPixelBus/pull/341
Let me know how this works for you.

feindsender, 2020-02-27:
Hi,
just did a quick test using WLED and it seems be fixed.
Thanks for the quick fix and also for your awsome library.

greetings
Maik

voyagertm, 2020-02-28:
I did a quick test with WLED also.
It is completly fixed now, thank you very much Makuna!!

Greetings Tony

Def3nder, 2020-03-01:
Hi @Makuna, someone on discord did try the latest master with WLED (where the flashing had been when fading brightness) and he did report that it works like expected now:

> Nevermind, used the latest master from makuna.
> 
> It works!!!
> Flashing is completly gone during boot and brightness changes.
> I am a very happy person right now :grin: :grin: :grin:

Do you know when you release 2.5.7 with this included ?
Great work ! I like the library !

Makuna, 2020-03-01:
v2.5.7 just made; might take a day to make it through the Arduino update process.

closed 2020-03-01 by Makuna

Def3nder, 2020-03-02:
Thanx a lot ! I appreciate your support 😃