Makuna/NeoPixelBus #346: SM16716 support possible?

ivoh95, 2020-04-03:
Any chance support for the SM16716 could be implemented? Would it just require adding a new method for that chip? Looks like this chip has clock and data and expects 50 0bits, followed by 1 1bit, followed by 3bytes for RGB values. Followed by another 1bit and the next pixels RGB values. 

So each pixels ends up being 25 bits, which is a leading 1 with 3x 8bits of RGB. 

50 0bits is a reset  command for the strip.

Atleast that's what I was able to make from online documentation.  Is it possible for this to be implemented seeing as how it doesn't send whole bytes?

ivoh95, 2020-04-10:
I gave this a shot myself and created a pull request. This is my first time doing anything like this, so there may be some mistake i didnt catch. Tested with a atmega328p, esp8266, and esp32 and a single sm16716 pixel and it seemed to work fine. Sadly i didnt have more than one pixel to test with. Please review the pull request and see if it seems like a good solution. Thanks! #349

Makuna, 2023-04-09:
https://github.com/Makuna/NeoPixelBus/pull/682

Makuna, 2023-04-11:
v2.7.4