Makuna/NeoPixelBus #348: P9813 Clock edge bug

PRO2XY, 2020-04-07:
**Describe the bug**
I am testing the P9813 support on your library. I found that the data is supposed to be clocked in on the rising clock edge, whereas the code in [transmitByte](https://github.com/Makuna/NeoPixelBus/blob/0177b8d5193f98541c08c461f6b59b789ed8fca3/src/internal/TwoWireBitBangImple.h#L71) clocks it in on falling edge. 

Changing the code to `digitalWrite(_pinClock, LOW);` before `digitalWrite(_pinClock, HIGH);` solves the bug with P9813, but I have not tested the implications it may have on other chipsets.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile and upload for P9813 chipset.
2. See random flickering on the LEDs

**Expected behavior**
LEDs behaving properly.

**Development environment (please complete the following information):**
 - OS: Win10
 - Platformio
 - Board target ESP8266 Arduino
 - Latest on master as of posting

**Additional context**
I referred to the ChainableLED library for figuring out the correct clock edge. 
https://github.com/pjpmarques/ChainableLED/blob/8764e66f86ab39bea38f3a87ac47d6a5a6fba136/ChainableLED.cpp#L61

Makuna, 2020-04-08:
What line in what code file are you changing?  You can include a link like...

https://github.com/Makuna/NeoPixelBus/blob/a9a031f10e7d3c9584cd5218c240206129269309/src/internal/TwoWireBitBangImple.h#L71

I suspect this is it.

This sets the data, the clock is brought high and then low, so the data is present on both the rising and failing edge of the clock.  

The spec states:
"The ic will latch a bit of data when the rising edge of the clock coming, And the data
should changed after the falling edge of the clock;"

My best translation/understanding of this is that it reads the data on the rising edge and the data pin should retain state until after the clock pin is lowered.
The spec isn't precise and clear as it doesn't include the standard timing charts.

@feindsender (author) Can you look into this?  Did it work for you?

feindsender, 2020-04-08:
Hi,
just tested with P9813Method and P9813SpiMethod both are working fine in my setup.
I am using Wemos D1 ESP8266 and 3 P9813 modules.
greetings
Maik

Makuna, 2020-04-08:
Are you running at 80mhz or 160mhz on that esp8266?  (its an option in the tools menu, default is 80mhz, but it runs only a little warmer at 160mhz).

feindsender, 2020-04-08:
Just tested with 160mhz and no problems.
The only thing i could imagine is the setting of pin3 of the p9813 chip.
On my boards this pin is high.
But the datasheet is not very clear about what this pin does.
`JEN=1,timing no parity,JEN=0,timing verification,drive mode,built-in pull-up `

@PRO2XY can you check if your pin3 is high or low? Also which board are you using?

PRO2XY, 2020-04-08:
I'm using Platformio, so not exactly sure if its 80MHz or 160MHz. 

P9813 boards I'm using were [sourced from Aliexpress](https://www.aliexpress.com/item/32858168662.html).
https://www.aliexpress.com/item/32858168662.html

Pin 3 on the P9813 is at 2.36V (doesn't sound very good).

Without making the modification I mentioned in OP, the LEDs would just flicker white every time data was sent to them. After making the mod the LEDs are working very well. I used this in the WLED project, paired it up with LEDFX, and I'm in music reactive lighting bliss right now :-D

Side note: if anyone is planning to buy these boards, I'd recommend not to. They seem to use just about any mosfet on the output with a lot of variation, and some of these have too high Vgs to work reliably.

feindsender, 2020-04-08:
I am using the same boards also from a different seller. 
My boards are working without a problem and i am also using WLED.
On mine i get 3.3v on pin3 when powering the boards with 3.3v.

PRO2XY, 2020-04-08:
Consider this solved. 

@feindsender getting 3.3V on pin3 got me thinking in terms of a power supply issue. And indeed the 3.3V input pin on the very first module had a solder crack. 3.3V rail was running at ~2.4V (This probably explains my mosfets not being driven properly)

I soldered the pin properly, reverted my changes in the library, and uploaded it again, and indeed the LEDs work exactly as expected. 

Assembly quality of some of these boards seems to be quite dodgy.

closed 2020-04-08 by Makuna