Makuna/NeoPixelBus #351: Will not compile with warnings severe enough to be treated as errors

marcmerlin, 2020-05-18:
In arduino preferences, I have "warnings: all" because it's the only way to find so many programming errors. In a few libraries, it gives warnings that can be cleaned up.
With NeoPixelBus, the warnings are severe enough that the compiler treats them as errors and refuses to build.
I can indeed turn them all off and then it builds, but basically I'm building code full of pretty bad warnings.

I'm using arduino 1.8.9 building for ESP32 using neopixelbus 2.5.7.

```
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:79:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:88:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:103:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:112:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:125:119: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     __attribute__((noinline)) static const void ICACHE_RAM_ATTR send_pixels(uint8_t* pixels, uint8_t* end, uint8_t pin)
                                                                                                                       ^
In file included from /tmp/arduino_build_684949/sketch/FX.h:30:0,
                 from /tmp/arduino_build_684949/sketch/fcn_declare.h:79,
                 from /tmp/arduino_build_684949/sketch/wled.h:88,
                 from /tmp/arduino_build_684949/sketch/xml.cpp:1:
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In constructor 'NeoPixelWrapper::NeoPixelWrapper()':
NpbWrapper.h:326:53: error: 'NeoPixelWrapper::_pGrbw' will be initialized after [-Werror=reorder]
   NeoPixelBrightnessBus<PIXELFEATURE4,PIXELMETHOD>* _pGrbw;
                                                     ^
NpbWrapper.h:322:16: error: 'NeoPixelType NeoPixelWrapper::_type' [-Werror=reorder]
   NeoPixelType _type;
                ^
NpbWrapper.h:157:3: error: when initialized here [-Werror=reorder]
   NeoPixelWrapper() :
   ^
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In member function 'void NeoPixelWrapper::Begin(NeoPixelType, uint16_t)':
NpbWrapper.h:176:12: error: enumeration value 'NeoPixelType_None' not handled in switch [-Werror=switch]
     switch (_type)
            ^
NpbWrapper.h:176:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In member function 'void NeoPixelWrapper::Show()':
NpbWrapper.h:266:12: error: enumeration value 'NeoPixelType_None' not handled in switch [-Werror=switch]
     switch (_type)
            ^
In file included from /home/merlin/Arduino/libraries/NeoPixelBus/src/NeoPixelBus.h:96:0,
                 from /home/merlin/Arduino/libraries/NeoPixelBus/src/NeoPixelBrightnessBus.h:29,
                 from /tmp/arduino_build_684949/sketch/NpbWrapper.h:144,
                 from /tmp/arduino_build_684949/sketch/FX.h:30,
                 from /tmp/arduino_build_684949/sketch/fcn_declare.h:79,
                 from /tmp/arduino_build_684949/sketch/wled.h:88,
                 from /tmp/arduino_build_684949/sketch/wled_serial.cpp:1:
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:79:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:88:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:103:60: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void setPin(const uint32_t pinRegister)
                                                            ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:112:62: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  inline const static void resetPin(const uint32_t pinRegister)
                                                              ^
/home/merlin/Arduino/libraries/NeoPixelBus/src/internal/NeoEspBitBangMethod.h:125:119: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     __attribute__((noinline)) static const void ICACHE_RAM_ATTR send_pixels(uint8_t* pixels, uint8_t* end, uint8_t pin)
                                                                                                                       ^
NpbWrapper.h:266:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
/tmp/arduino_build_684949/sketch/NpbWrapper.h:265:10: warning: unused variable 'b' [-Wunused-variable]
     byte b;
[... 547 lines cut ...]
NpbWrapper.h:313:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
/tmp/arduino_build_684949/sketch/NpbWrapper.h: In member function 'void NeoPixelWrapper::cleanup()':
NpbWrapper.h:330:12: error: enumeration value 'NeoPixelType_None' not handled in switch [-Werror=switch]
     switch (_type) {
            ^
NpbWrapper.h:330:12: error: enumeration value 'NeoPixelType_End' not handled in switch [-Werror=switch]
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
'NeoPixelWrapper::_pGrbw' will be initialized after [-Werror=reorder]
```

Makuna, 2020-05-18:
many of the ones listed are not from the library, but are from some sort of wrapper.

While I agree, there have been issues of compatibility between platforms on support and base types, like what type is a pin id?  The ability to correctly apply attributes to static member functions.  In these cases there will be a warning to continue to track the incompatible.  But some can be addressed.

marcmerlin, 2020-05-20:
my apologies, you are correct, none of those errors are actually in neopixelbus. I'll close this and just moved the bug to https://github.com/Aircoookie/WLED/issues/929

closed 2020-05-20 by marcmerlin