Makuna/NeoPixelBus #357: WS2816 support

GerwinJanssen, 2020-06-16:
Is possible to add support for WS2816 a 16 bit digital LED?

[WS2816C-2121 Datasheet_EN_V1.0.pdf](https://github.com/Makuna/NeoPixelBus/files/4784346/WS2816C-2121.Datasheet_EN_V1.0.pdf)

mcer12, 2022-06-13:
These chips seem to be quite cheap on lcsc, is there any hope this will be implemented or is this not a priority (this year)? @Makuna

Makuna, 2022-06-13:
https://github.com/Makuna/NeoPixelBus/pull/581

Previous ground work was done already to expose a Rgb48Color object and color features to use them.  

the above merge adds NeoGrb48Feature and specifically named NeoGrb48Ws2816Feature (they are really the same thing so either will work for you)

The method is the same as Ws2812x, so you can use the Neo800KbpsMethod. 

Leaving this open to track adding a specific method NeoWs2816Method (will be a typedef like Neo800KbpsMethod).

mcer12, 2022-06-13:
@Makuna Thank you, will try it out :) Maybe add the WS2816 to supported chips? There's no mention of it in wiki (or anywhere on the internet really), I think the chip is interesting enough.

Makuna, 2022-06-13:
https://github.com/Makuna/NeoPixelBus/pull/582
added methods specific to WS2816.

alexgubanow, 2023-01-21:
hello, im struggling to get the ws2816b working nicely with NeoPixelBus:
- only full colors are shown, no shady things
- white is not white
How can i debug this thing?

Makuna, 2023-01-21:
@alexgubanow Best practices for debugging wonky NeoPixelBus issues.  

Have a logic analyzer to attach to the output so you examine the signal directly.  But most people wont have one, so...

1) Understand what platform you are using (Avr?  ESP8266?  ESP32?) and always include this when starting a discussing with others.
2) Also, Always include how you define your NeoPixelBus when starting. 
3) Start with the simplest of sketches (like the included NeoPixelTest, and of course modify as needed for your Bus definition and color type used)
4) Be specific when describing what you see, like "I set the zero pixel to Rgb48Color(6000,0,0) but it displays as bright blue"

(p.s.  I am on discord (Makuna) at the moment and since this is a LED I actually don't have and I am waiting for feedback on this pull, I am willing to walk you through some things proactively from there)

alexgubanow, 2023-01-21:
ok, sorry, i didn't expect any answer so haven't described anything in details.
I do have picoscope 3204D, should be good enough to see the signal, if need.
I do have arduino nano and esp8266 and esp32, but targeting nano for this project.
i do use NeoPixelTest as base, colors are presented as expected, maybe bit more to the bright side from regular.
For example, im not able to get purple, it gives me or pink or blue, nothing in the middle.

`#include <NeoPixelBrightnessBus.h>

const uint16_t PixelCount = 58; // this example assumes 4 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 2;  // make sure to set this to the correct pin, ignored for Esp8266

#define colorSaturation 65535

// three element pixels, in different order and speeds
NeoPixelBrightnessBus<NeoGrb48Feature, NeoWs2816Method> strip(PixelCount, PixelPin);

Rgb48Color red(colorSaturation, 0, 0);
Rgb48Color green(0, colorSaturation, 0);
Rgb48Color blue(0, 0, colorSaturation);
Rgb48Color white(colorSaturation);
Rgb48Color black(0);

Rgb48Color Fuchsia(65535, 0, 65535);
Rgb48Color NavyBlue(14000, 0, 30000);

Rgb48Color b1 = Rgb48Color::LinearBlend(Fuchsia, NavyBlue, 0.75);
Rgb48Color b2(102, 0, 102);

void setAllLeds(Rgb48Color targetColor)
{
  for(uint16_t led = 0; led < PixelCount; led++)
  {
    strip.SetPixelColor(led, targetColor);
  }

}

void setup()
{
    Serial.begin(115200);
    while (!Serial); // wait for serial attach

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    setAllLeds(black);
    strip.SetBrightness(0);
    strip.Show();
}
uint16_t currLed = 0;
uint16_t color = 0;
void loop()
{
    delay(30);
    setAllLeds(NavyBlue);
    Rgb48Color calc(0, 0, color);
//    setAllLeds(calc);
//setAllLeds(white);
    // strip.SetPixelColor(3, Fuchsia);
    // strip.SetPixelColor(4, Fuchsia);
    // strip.SetPixelColor(5, Fuchsia);
    // strip.SetPixelColor(6, b1);
    // strip.SetPixelColor(7, b1);
    // strip.SetPixelColor(8, b1);
    // strip.SetPixelColor(9, NavyBlue);
    // strip.SetPixelColor(10, NavyBlue);
    // strip.SetPixelColor(11, NavyBlue);

    strip.SetPixelColor(currLed, white);
    strip.SetBrightness(10);
    strip.Show();
    currLed = currLed < PixelCount ? currLed + 1 : 0;
    color = color < colorSaturation ? color + 1 : 0;
}`

alexgubanow, 2023-01-21:
matroskin_1307#1664 me in discord

Makuna, 2023-01-21:
We ran into this https://github.com/Makuna/NeoPixelBus/issues/622, otherwise seems to be working.

Makuna, 2023-02-03:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.2

closed 2023-02-03 by Makuna

tablatronix, 2023-05-05:
Has anyone tested this or have it working ?  The bit order on my scope looks wrong like each byte is LSB->MSB and not for all 16bits in order ? Will try to get find my logic analyzer for better capture

Makuna, 2023-05-05:
its considered bad form to post to a closed issue.  You should use discussions first to start the conversation.

CHARACODES, 2024-01-25:
How to control ws2816c led using python in RPI 4

CHARACODES, 2024-03-07:
Hi, looking support to develop driver for the WS2816 series RGB LED using python for Pi compute module.

Makuna, 2024-03-07:
@CHARACODES : It is considered bad form to post to a closed issue.  Especially when its completely unrelated to this repo.  

You ask questions in the discussion area.