Makuna/NeoPixelBus #361: Documentation: re: FastLED and RMT

bbulkow, 2020-08-17:
This page: 
https://github.com/Makuna/NeoPixelBus/wiki/Library-Comparisons

States:
Interrupt driven on ESP8266, so it's sensitive to timing (not an issue on ESP32 which uses RMT)

It turns out the ESP32 and RMT and FastLED has a rather involved situation regarding interrupts. The FastLED code isn't located in IRAM, thus a large number of operations ( like serving network responses ) will interrupt the feeding of the RMT driver, making it susceptible to flashing. This is under active investiagion in FastLED as of 3.3, but hasn't been resolved. 

While it wouldn't be reasonable to slag a competing library, your information that this is "not an issue on esp32" is simply false, and I'd suggest saying just "senstive" for both ESP8266 and ESP32.

Makuna, 2020-08-17:
Thanks for catching this.  This comment is in the wiki is old information.  
There were several versions of the core that this was completely true; but an update to the core last year caused RMT to have several issues, also being tracked for NeoPixelBus.

bbulkow, 2020-08-17:
Are you saying that NeoPixelBus does, or might, suffer from the same flashing issues as FastLED does?

Makuna, 2020-08-17:
take a look now and provide feedback.

Makuna, 2020-08-17:
There is a core issue right now on Esp32 that if you have some feature that is causing a high frequency of interrupts (60+ a second) and time to service those ISRs is long enough, it will starve the RMT ISR causing irregular pulse generation.  It was thought this might be due to the IRAM_ATTR not being applied correctly ( GCC bug); but hacking it so it is has proven not to solve the issue.

bbulkow, 2020-08-17:
Thanks, that update looks lovely.

closed 2020-08-17 by bbulkow