Makuna/NeoPixelBus #368: LPD8806 compilation error

Aircoookie, 2020-09-20:
Hi @Makuna , unfortunately there seems to be an issue with LPD8806 support currently.
For reference, here is the [related issue](https://github.com/Aircoookie/WLED/issues/971) in my project, which uses your excellent library!

**Describe the bug**
Basic LPD8806 example fails to compile.

**To Reproduce**
Steps to reproduce the behavior:
1. Use provided example `NeoPixelTest.ino`
2. Use bus definition `NeoPixelBus<Lpd8806GrbFeature, Lpd8806Method> strip(PixelCount, 2, 3);`. The hardware SPI variant `NeoPixelBus<Lpd8806BrgFeature, Lpd8806SpiMethod> strip(PixelCount);` produces the same issue.
3. Compile and see error regarding the `Lpd8806GrbFeature`. Similar error occurs with `Lpd8806BrgFeature`.

**Expected behavior**
LPD8806 support to compile without errors

**Development environment**
 - OS: Win 10
 - Build Environment: reproduced both in Arduino IDE v.1.8.9 and latest platform.io
 - Board target: reproduced for Arduino Uno, ESP8266 core versions v.2.6.3 and v.2.7.4
 - Library version: reproduced both in v.2.5.7 and latest master (commit a05b708)

**Additional context**
Compilation error provided below:
```
In file included from C:\Users\cschwinne\Documents\Arduino\NeoPixelTestLPD8806\NeoPixelTestLPD8806.ino:14:0:

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h: In instantiation of 'class NeoPixelBus<Lpd8806GrbFeature, Lpd8806MethodBase<TwoWireBitBangImple> >':

C:\Users\cschwinne\Documents\Arduino\NeoPixelTestLPD8806\NeoPixelTestLPD8806.ino:52:53:   required from here

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h:363:10: error: no type named 'SettingsObject' in 'class Lpd8806GrbFeature'

     void SetPixelSettings(const typename T_COLOR_FEATURE::SettingsObject& settings)

          ^

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h: In instantiation of 'NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::NeoPixelBus(uint16_t, uint8_t, uint8_t) [with T_COLOR_FEATURE = Lpd8806GrbFeature; T_METHOD = Lpd8806MethodBase<TwoWireBitBangImple>; uint16_t = short unsigned int; uint8_t = unsigned char]':

C:\Users\cschwinne\Documents\Arduino\NeoPixelTestLPD8806\NeoPixelTestLPD8806.ino:52:69:   required from here

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h:135:106: error: 'SettingsSize' is not a member of 'Lpd8806GrbFeature'

         _method(pinClock, pinData, countPixels, T_COLOR_FEATURE::PixelSize, T_COLOR_FEATURE::SettingsSize)

                                                                                                          ^

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h: In instantiation of 'uint8_t* NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::_pixels() [with T_COLOR_FEATURE = Lpd8806GrbFeature; T_METHOD = Lpd8806MethodBase<TwoWireBitBangImple>; uint8_t = unsigned char]':

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h:225:54:   required from 'void NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::SetPixelColor(uint16_t, typename T_COLOR_FEATURE::ColorObject) [with T_COLOR_FEATURE = Lpd8806GrbFeature; T_METHOD = Lpd8806MethodBase<TwoWireBitBangImple>; uint16_t = short unsigned int; typename T_COLOR_FEATURE::ColorObject = RgbColor]'

C:\Users\cschwinne\Documents\Arduino\NeoPixelTestLPD8806\NeoPixelTestLPD8806.ino:95:31:   required from here

C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h:391:57: error: 'pixels' is not a member of 'Lpd8806GrbFeature'

         return T_COLOR_FEATURE::pixels(_method.getData());

                                                         ^

Multiple libraries were found for "NeoPixelBus.h"
 Used: C:\Users\cschwinne\Documents\Arduino\libraries\NeoPixelBus-master
 Not used: C:\Users\cschwinne\Documents\Arduino\libraries\arduino_689834
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

Thank you for your time and dedication to this library, it truly is the best option for driving addressable LEDs!

Makuna, 2020-09-20:
https://github.com/Makuna/NeoPixelBus/pull/369

Makuna, 2020-09-20:
@Aircoookie Let me know if you spot anything else.  I will be releasing a new build very soon.

Aircoookie, 2020-09-20:
@Makuna thank you for the lightning fast fix!
Compiles without issues now, still waiting on community feedback whether the LEDs behave correctly as I don't personally have LPD8806. For now I believe we can close this though. Awesome work!

Makuna, 2020-09-20:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.6.0

closed 2020-09-20 by Makuna