Makuna/NeoPixelBus #372: Changing the interrupt level on ESP32

programminghoch10, 2020-09-23:
**Problem**
I have a project where the LED-Strip is the main component.
The LED-Strip of my project also glows in the night (very darkly) and any flicker will be noticed instantly by the user.
WiFi connectivity and a webserver are only secondary, and I will thus give up wifi and webserver packets in exchange for good/perfect LED performance.

**Potential Solution**
I noticed #321 and until that issue is permanently solved and we always get perfect results, it would be nice to set a higher interrupt level myself in order for RMT filling interrupts to always be handled before wifi/webserver interrupts.

I propose addition of a define directive to set the interrupt level before including NeoPixelBus to my project.

Makuna, 2020-09-23:
Did you try changing the interrupt and seeing if it fixed your issue?

I believe it was tried and it did nothing.

programminghoch10, 2020-09-24:
I was only able to increase the interrupt level to 3, as anything beyond triggered an error upon boot (not during compile though):
```
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4008a858
file: ".pio/libdeps/env/NeoPixelBus/src/internal/NeoEsp32RmtMethod.h" line 399
func: void NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::Initialize() [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel6]
expression: rmt_driver_install(T_CHANNEL::RmtChannelNumber, 0, ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_LEVEL4)
```
LEDs were still flickering at level 3.

I also tried [this](https://github.com/Makuna/NeoPixelBus/issues/321#issuecomment-571343639) comment by @BlueAndi as far as I could replicate his explanation, but it did not change a bit. 
I also do not understand how locking onto the thread calling `Show()` should help the interrupt in any way.

Makuna, 2020-09-24:
That error is a parameter check in the IDF, you can't use level4+ for the call to attach to this interrupt.

Your findings are what I was stating, changing the interrupt level does not improve the issue.

Are you using my latest release (or github as it was only made days ago)?  The newest release includes the changes to work around the RMT translate not being in IRAM.  The github has had this change for weeks.

programminghoch10, 2020-09-24:
Yes I am using this exact version, it still flickers though.

I've also tried I2s, its flickers more, but also more consistently and regurarily and does not get influenced by a webserver or SPIFFS read/write.

Do you have any ideas on how I2s could be improved, even with drawbacks to give perfect results?

Makuna, 2020-09-24:
Right now, the only solution is to reduce network traffic.  At some point something changed in the core that made both RMT and I2S less reliable when traffic was high and there is nothing I can do about.  

The Github of Esp32 Arduino uses a newer IDF which was better (I have a pending pull that works with it but not the current release) that will improve the RMT call back code size.

Makuna, 2020-09-24:
This issue was tracking increasing the IRQ, which has shown to not solve anything.  So I will close it soon.  

The issues area is for bug tracking not conversations, jump on gitter to ask questions.

closed 2020-09-24 by programminghoch10

Makuna, 2020-09-24:
@programminghoch10 
BlueAndi commented on Gitter today....
"I am using ESP SDK v3.2.3-14-gd3e562907, Arduino IDF release/v3.2, LwIP 2.0.3 and NeoPixelbus head + RMT ISR prio 2. Additional I block the webserver till the matrix is updated too, as I mentioned in one of the issues."

Makuna, 2020-09-27:
BlueAndi commented on gitter today...
"Tested now RMT prio1 and prio2. One test case with iperf (tcp) and one with active websocket connection transfering periodically data.
I couldn't see any difference between prio1 and prio2. Interesting was that with running iperf there were really no artifacts, but with the websocket connection running there were isolated artifacts.
iperf uses a LwIP socket connection, running on APP_CPU with prio 10. The websocket connection is running via AsyncTcp, running on APP_CPU with prio 3. The task which updating the display runs on APP_CPU with prio 3 (to run round-robin with AsyncTcp) and its submitting computing time for tasks <= prio 3 only after the display update finished (via CanShow()).

So long answer, short result: prio1 seems to be enough"

programminghoch10, 2020-10-08:
I still do not understand what he meant with 
`after the display update finished (via CanShow())` .
Does he mean waiting for finish like this?
```cpp
strip.Show();
while (!strip.CanShow()) ;
```

Anyhow is there a possibility to have NeoPixelBus lock onto the task calling `Show()` and using it for output?
Maybe this way we can prevent flicker, by just dedicating one core/task with high priority for the led output.

Makuna, 2020-10-08:
@BlueAndi @programminghoch10 Programming is looking for a more detailed answer.  Do you have a link to your code he can glance at?

BlueAndi, 2020-10-09:
See my task here, which is responsible to start updateing the matrix:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/a0eb82c502455f0e4fe7654f8b4244215d45637d/src/Gfx/DisplayMgr.cpp#L877-L921

Especially with the line:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/a0eb82c502455f0e4fe7654f8b4244215d45637d/src/Gfx/DisplayMgr.cpp#L904
which calls
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/a0eb82c502455f0e4fe7654f8b4244215d45637d/src/Gfx/LedMatrix.h#L100-L108

For a bigger overview, see my design here:
https://github.com/BlueAndi/esp-rgb-led-matrix/blob/master/doc/design/gfx.png

Hope this helps and gives more detail. A more detailed discussion can be done via gitter, so we avoid spamming the issue. ;-)