Makuna/NeoPixelBus #375: NeoPixelBrightness Example bug

tablatronix, 2020-10-06:
setBrightness is pixel destructive.
But the example loop does not reset the pixel colors, so it only fades once then just sits there dimming black up and down.

Need to re-set colors in the loop before set brightness to get a proper fade down up example.

tablatronix, 2020-10-06:
```C++
    // apply dimming
    brightness += direction;

    strip.SetPixelColor(0, red);
    strip.SetPixelColor(1, green);
    strip.SetPixelColor(2, blue);    
    strip.SetBrightness(brightness);
```

I dont have to time to PR sorry

Makuna, 2020-10-06:
I consider this less a bug and more of a sample improvement.    Its a good suggestion.

In general, I do not recommend the use of the NeoPixelBrightnessBus due to the destruction nature and the nuances of the effects.

closed 2023-05-01 by Makuna