Makuna/NeoPixelBus #376: BitBang latch delay missing ?

tablatronix, 2020-10-06:
ESP Core latest stable 
Lib latest stable

**Describe the bug**
Bit Bang on esp8266 show latch missing or no delay, multiple show() will get shifted and latched together

**To Reproduce**
Based on example code

Set a code with a strip of 1 pixel, connect MORE than 1 pixel.

```C++
    strip.Begin(); // no show to easier show bug ( else its black )
    strip.SetPixelColor(0, red);
    strip.Show();
    strip.SetPixelColor(0, green);
    strip.Show();
```

**Expected behavior**
pixel 0, should be green
Pixel 1 should stay off.

** Observed**
Pixel 0 is green, pixel 1 is red

**Development environment (please complete the following information):**
pio

**Minimal Sketch that reproduced the problem:**
```c++
void setup()
{

#include <NeoPixelBrightnessBus.h> // instead of NeoPixelBus.h

const uint16_t PixelCount = 1; // this example assumes 3 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 12;  // make sure to set this to the correct pin, ignored for Esp8266

#define colorSaturation 255 // saturation of color constants
RgbColor red(colorSaturation, 0, colorSaturation);
RgbColor green(0, colorSaturation, colorSaturation);
RgbColor blue(0, 0, colorSaturation);

// Make sure to provide the correct color order feature
// for your NeoPixels
NeoPixelBrightnessBus<NeoRgbFeature, NeoEsp8266BitBang800KbpsMethod> strip(PixelCount, PixelPin);

    Serial.begin(115200);

    Serial.println();
    Serial.println("Initializing...");

    strip.Begin();
    // strip.Show();
    Serial.println();
    Serial.println("Running...");

    strip.SetPixelColor(0, red);
    strip.Show();
    strip.SetPixelColor(0, green);
    strip.Show();

    delay(5000);
}

void loop(){}
```

Makuna, 2020-10-06:
How do you know the delay is missing?  Did you put it on a scope or logic analyzer? Relying on the correct color is not good enough, especially with bitbang on Esp32 or Esp8266.

The bitbang method should not be used on either of these platforms.  Due to the interactions with WiFi (interrupts and RTOS) and the inability to guard against it the data stream can be interrupted.

https://github.com/Makuna/NeoPixelBus/wiki/ESP8266-NeoMethods
https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods

The code does time the delay using micros().

tablatronix, 2020-10-06:
I can scope it, but do not need to. This isn't wrong color its pixels, extra pixels being shifted when show is called twice. 

Also its not interrupt or glitches, I am very familiar with esp chips. 

Its easy enough to reproduce. 

Ill check the timing and code when I get a chance.

tablatronix, 2020-10-06:
There is definitely a delay for reset, How do I specify the other class speeds , I have tried a bunch of pixels, maybe I need longer resets.

`NeoEspBitBangSpeedWs2812x` ?

tablatronix, 2020-10-07:
ok well it looks like, none of my pixels actually reset at 60us, they seem to all need 300us..
I guess that is why the adafruit library always works for everything.

I guess I will close this, I did not realize you had all datasheet specs reset times for all types

Ill have to test them all and see if any actually latch at 50-80us

closed 2020-10-07 by tablatronix

Makuna, 2020-10-07:
These are the primary ones with different reset times.  Many LEDs maybe sold as one but are actually another.

800Kbps = 50us
Ws2812x = 300us
Sk6812 = 80us
Tm1814 = 200us