Makuna/NeoPixelBus #377: get RgbColor as uint32_t

logowe, 2020-10-12:
I need to store several colours set in a session in order to have them automatically read back at the start of the next session.

A cast uint32_t(RgbColor) is not be successful because RgbColor is no standard type.

A new function like this could yield the colour needed in a uint32_t type value:

uint32_t GetColor(RgbColor NewColour)

Or maybe I missed something which would be feasable using the current functionality of the library?

Makuna, 2020-10-12:
please use the gitter channel  to ask questions as the readme and wiki request.  

it cant be cast as it is 3 bytes not four.  you have cast to pointer to byte and copy.  there should be no need to have the extra method.  Paste the use case in gitter and a response on how to do it will follow.

Makuna, 2020-10-18:
Discussed and I believe you understand that this request is not required as the RgbColor can directly be serialized as three bytes.

closed 2020-10-18 by Makuna