Makuna/NeoPixelBus #379: ESP32-S2 (Arduino IDE): error: 'i2s_dev_t' has no member named 'pdm_conf'

Miraculix200, 2020-10-25:
I'm trying to use NeoPixelBus on a ESP32-S2 using the esp32s2 tree of the Arduino IDE (not ESP-IDF) from here:
https://github.com/espressif/arduino-esp32/tree/esp32s2

During compilation I get this error:
NeoPixelBus_by_Makuna\src\internal\Esp32_i2s.c:334:10: error: 'i2s_dev_t' {aka 'volatile struct <anonymous>'} has no member named 'pdm_conf'; did you mean 'pd_conf'?

Renaming pdm_conf to pd_conf leads to other errors (I have no idea what I'm doing).

I guess the ESP32-S2 support of this library was written for ESP-IDF, and that may be the problem. The esp32s2 tree of the Arduino IDE is supposed to use mostly the same libraries as ESP-IDF however. If I understood it correctly, the esp32s2 tree uses a later version of the SDK than ESP-IDF: SDK version v4.3-dev-907-g6c17e3a64-dirty

Is there a quick way to fix this? E.g. disabling I2S support.

Makuna, 2020-10-25:
please follow the template.
provide versions:
* of the esp32s2 branch (date/time at least).
* of this library

also, above the error line, include a line that compiles such that you can show what symbols are being used.  A line that is usually VERY long which includes "-D blah" repeated many times.

Miraculix200, 2020-10-25:
Version of the esp32s2 branch: cloned on September 10, 2020 (head 2656452d044215d5a49ba6e909f4ea19c089baa9 I guess)
Version of this library: 2.6.0

I'm not sure which line you mean as there are several lines with -D parameters, so I copied the entire output of the compiler: https://gist.github.com/Miraculix200/1b92564b4afdb66464a1d078a98cbf21

Makuna, 2020-10-25:
They seemed to have changed how things have built since I last added the support.

search for CONFIG_IDF_TARGET_ESP32S2 and replace with ARDUINO_ESP32S2_DEV and see if that fixes it for you.

Miraculix200, 2020-10-26:
I've replaced it in esp32_i2s.c, NeoEsp32I2sMethod.h and NeoEsp32RmtMethod.h, and it still fails to compile with the same error:
https://gist.githubusercontent.com/Miraculix200/4417819599e9b180ced6419db12272b8/raw/6b7b458df2b67bf4f6a5551db365591353d102ff/gistfile1.txt

Makuna, 2020-10-26:
bigger changes to is2 than I would have thought between releases.  There is no PDM settings exposed on ESP32S2 now.

You will have to wrap all those statements with an ifdef like

```
#if !defined(CONFIG_IDF_TARGET_ESP32S2)
    i2s->pdm_conf.rx_pdm_en = 0;
    i2s->pdm_conf.tx_pdm_en = 0;
#endif
```

One I can reset the ESP32S2 environment again, I will apply and test required changes; but it won't be for at least a few days.

Miraculix200, 2020-11-13:
Ok, I've tried it now. Just letting you know it works. But I guess you knew it would work anyway.

Makuna, 2020-11-13:
https://github.com/Makuna/NeoPixelBus/pull/382

Miraculix200, 2020-11-13:
Actually I did it like this, because a few lines below it is using that statement too:
#if !defined(ARDUINO_ESP32S2_DEV)

Makuna, 2020-11-13:
ARDUINO_ESP32S2_DEV isn't present in the code; why does yours have this anywhere?

Miraculix200, 2020-11-13:
Oh I see. I guess I changed that (in line 375) and forgot about it. And then copypasted it today

Miraculix200, 2020-11-13:
I'll try compiling it with the changes you made

Miraculix200, 2020-11-13:
All still ok with #if !defined(CONFIG_IDF_TARGET_ESP32S2)

Makuna, 2021-01-31:
2.6.1 released

closed 2021-01-31 by Makuna

rahmanshaber, 2021-04-16:
Is ESP32-s2 fully supported? 
i tried to compile the NEOpixel_brightness demo, it does not compiles. Got "Neo800KbpsMethod was not declared". But it compiles fine for Arduino Micro borad.

Makuna, 2021-04-16:
@rahmanshaber Please use the discussions tabs for questions like this.  It is considered bad form to reply to a closed issue unless it clarifies the issue more.