Makuna/NeoPixelBus #383: Document suggestion: UART FIFO size implications

matthijskooijman, 2020-11-14:
I've been looking for a non-blocking (usable from an ISR) neopixel library on the ESP8266, and it looks like this is a great one. 

However, it is not entirely clear from the documentation to what degree the sync UART methods will block. The documentation suggests to me that the Async UART and DMA methods will be entirely non-blocking, running entirely in hardware or interrupts, but for the Sync UART method, it only says "These methods use the CPU to manage a small hardware-managed UART buffer to send the data to the NeoPixels. Thus, it requires more CPU overhead than the DMA method, but it does NOT require an extra buffer."

Looking at the code, IIUC it will delegate the sending of data to the UART hardware by filling up the UART hardware FIFO as far as possible and if that is not enough to hold data for all pixels, *block* until more room becomes available. I believe this would be useful to document. I would add something to the wiki, but I'm not 100% I got this correct, so I'm leaving a suggestion below instead. If this looks good, let me know and I'll add it.

In the section https://github.com/Makuna/NeoPixelBus/wiki/ESP8266-NeoMethods#neoesp8266uart1800kbpsmethod--neoesp8266uart0800kbpsmethod I would suggest adding:

> Because the UART FIFO size is limited, data for just a few (typically 10) pixels can be sent completely in the background. If you have more pixels than will fit into the UART FIFO, data sending will block (use the CPU) until only a few pixels are left to send, which can then be completed in the background while the CPU does other things. The exact number of pixels depends on how much bytes each pixel needs. The FIFO buffer is 128 bytes, of which 4 bytes are need for each byte transmitted to the pixel, so for a typical 3-bytes-per-pixel WS2812 you need 12 FIFO bytes per pixel and can drive 10 pixels without blocking).

Makuna, 2020-11-14:
That looks good.

matthijskooijman, 2020-11-14:
Thanks for the feedback, I added it to the wiki :-)

closed 2020-11-14 by matthijskooijman