Makuna/NeoPixelBus #384: Arduino Nano 33 IoT Support

moooorph, 2020-12-04:
Arduino Nano 33 IoT (Arm® Cortex®-M0 32-bit SAMD21) support requested due to identical issues with Nano 33 BLE while using the onboard WiFi/BLE (u-blox NINA-W10) -> interrupt issues with standard methods, PWM requested (see _NeoNrf52xMethod.h_)

Tested with: NeoPixelBus 2.6.0 + Arduino Nano 33 IoT + _ArduinoBLE.h_ and standard Arm method (_NeoArmMethod.h_)
LED output deactivated: BLE working / LED output activated: BLE no connection possible  (interrupts/timing)

Makuna, 2020-12-04:
This will require specific hardware support for that platform.  The ARM support is a bitbang method and will be interrupted by lower level services like BLE and WIFI as you found out.

I do not own one of these boards myself.

Makuna, 2020-12-31:
@moooorph I got a Seeeduino XIAO for a holiday gift, so I am now actively working on it.  If you are interested in testing it on the Nano 33 IOT let me know.
I am working on i2s method first.  Then probably a i2c method.  The PWM feature looks heavy interrupt based so that will be the last method I will look into.

moooorph, 2021-01-09:
@Makuna excuse my delay. I decided to swap my project to an ESP32 platform (multiple reasons) and can use RMT now which is my most preferred solution acc. the described timing issues.
Anyway I could try to test your XIAO methods with the Nano IoT