Makuna/NeoPixelBus #388: Add support for "old" LPD6803 led strings

WerkAG, 2020-12-15:
I have some "old" LPD6803 led strings, and tried to make it work with WLED.
Based on some already supported LED types, I created two new files for the LPD6803:
Lpd6803ColorFeatures.h
Lpd6803GenericMethod.h

Both added to NeoPixelBus.h file

With minimal changes on the the NpbWrapper.h of WLED, I have the LPD6803 LED strings working well since almost 10 days (very occasionally there are some flicker on transitions between some kind of effects, which I believe is due to narrowing colours
 from 8 to 5 bits).

In case you think that adding support for those kind of LED strings can be useful, feel free to use attached files. Despite they are working for me, please first revise them, I'm not a programmer :-)

Merry Christmas

[LPD6803.zip](https://github.com/Makuna/NeoPixelBus/files/5699305/LPD6803.zip)

brianmiller1956, 2021-01-13:
I too have some "old" lpd6803 strings, I would also like the ability to drive these devices from a binary download, I have been unable to get the arduino ide download to compile yet, I know that Fastled does support these leds, Many thanks.

Makuna, 2021-01-15:
There are some bugs in the provided code, that will cause reading and writing past buffers.  I am working on fixing it

Makuna, 2021-01-15:
https://github.com/Makuna/NeoPixelBus/pull/402 has been merged in.

@brianmiller1956 @WerkAG 
It would be great to get some testing from someone who actually has some of these.

brianmiller1956, 2021-01-15:
Hi, yes I would be happy to do some testing with these leds, the ones I have are 12volt and have a 16 pin LPD6803 driving 3 RGB led packages, so my string of 150 leds has 50 chips in total, I did some testing yesterday with an arduino uno running Fastled and had to set the number of pixels to 50 to run every led but they did work well though I think that the leds are running in  groups of 3, I am a newcomer to arduino but have been in electronics and computing all my life, I have been using the binary flash of WLED, with no problems, I have downloaded the sketch version but have not yet succeeded in getting it to compile in the arduino IDE, 
I have 2 strings of 150 LPD6803 leds and will help in any way I can. 
Many thanks.

Makuna, 2021-01-15:
@WerkAG @brianmiller1956 

Sorry, another test please.  I found some bugs I introduced with endianness (byte order within a mulit-byte integral) and I JUST merged in those fixes.
https://github.com/Makuna/NeoPixelBus/pull/403

Please grab the latest. 

@brianmiller1956 Here is a simple sketch to test your config preset to 50 pixels, it will show a pattern of Red, Green, Blue, white, then Black; across the whole strip.  It will alternate every five seconds showing this pattern and clearing to black.  To get the latest of the lib, you can clone the repo into your arduino\libraries\NeoPixelBus directory, or download the zip and copy it to the same location.

```
#include <NeoPixelBus.h>

const uint16_t PixelCount = 50;

// make sure to set this to the correct pins
const uint8_t DotClockPin = 2;
const uint8_t DotDataPin = 3;  

// for software bit bang
NeoPixelBus<Lpd6803GrbFeature, Lpd6803Method> strip(PixelCount, DotClockPin, DotDataPin);

// for hardware SPI (best performance but must use hardware defined pins)
//NeoPixelBus<Lpd6803GrbFeature, Lpd6803SpiMethod> strip(PixelCount);

const uint8_t Saturation = 128;

RgbColor red(Saturation, 0, 0);
RgbColor green(0, Saturation, 0);
RgbColor blue(0, 0, Saturation);
RgbColor white(Saturation);
RgbColor black(0);

void setup()
{
    strip.Begin();
    // for ESP32, you need to do this instead if you want to set the pins as they support it
    // strip.Begin(DotClockPin, -1, DotDataPin, -1);
}

void loop()
{
    strip.ClearTo(black);
    strip.Show();
    
    delay(5000);

    // set the color pattern
    const uint16_t patternSize = 5;
    for (uint16_t pixel = 0; pixel < PixelCount; pixel += patternSize)
    {
      strip.SetPixelColor(pixel + 0, red);
      strip.SetPixelColor(pixel + 1, green);
      strip.SetPixelColor(pixel + 2, blue);
      strip.SetPixelColor(pixel + 3, white);
      strip.SetPixelColor(pixel + 4, black);
    }
    strip.Show();

    delay(5000);
}

```

WerkAG, 2021-01-15:
> @WerkAG @brianmiller1956
>
> Sorry, another test please. I found some bugs I introduced with endianness (byte order within a mulit-byte integral) and I JUST merged in those fixes.
> #403
>
> Please grab the [...]

WerkAG, 2021-01-15:
> it will show a pattern of Red, Green, Blue, white, then Black; across the whole strip. It will alternate every five seconds showing this pattern and clearing to black. 

Hi,
No problem with testing :-)

I have tried your sketch with latest changes, and I'm getting exactly the effect you said:   a pattern of Red, Green, Blue, white, then Black; across the whole strip and clearing to black every five seconds.

Tried latest changes with WLED, and seem everything is working fine too.
If you want / need any other test, just tell me.

Thanks.

Makuna, 2021-01-15:
great!  I will close this when I make the next official release of the library.

WerkAG, 2021-01-15:
Thank you for your work on this matter.  If you need any further test, just say.

Seems that accidentally I have closed this issue, sorry, I'm not used to GitHub

WerkAG, 2021-01-15:
Just adding that I tried it with an Wemos D1 mini. Right now, I haven't any ESP32 available.

brianmiller1956, 2021-01-16:
Hi, I have tested this sketch with an UNO and 150 led reel of 12 volt lpd6803, with pixelcount at 50 the whole reel lights, first 3 leds red, next 3 green, next 3 blue, next 3 white, then 3 black, repeating this to the end of the reel, whole reel then goes black and cycle repeats.
Pixelcount at 5 just lights the first 15 leds.
Colour sequence is correct so these are GRB.
Brian.

brianmiller1956, 2021-01-16:
Sorry I am telling a lie, the colour sequence is red, blue, green, white, black
So these are not grb they are BRG?

Makuna, 2021-01-16:
@brianmiller1956 This is what the feature is primarily for, the color order.  So then use the `Lpd6803BrgFeature` feature.

like...
```
NeoPixelBus<Lpd6803BrgFeature, Lpd6803Method> strip(PixelCount, DotClockPin, DotDataPin);
```

brianmiller1956, 2021-01-16:
Hi, I have altered this to BrgFeature the colour sequence is now correct as per the sketch, the reel shows 3 red, 3 green, 3 blue 3 white, 3 black leds. so it looks like the leds are grouped as 3 leds per pixel.
Thanks for your help.

WerkAG, 2021-01-16:
@brianmiller1956 Indeed on LPD6803 led strings each chip controls 3 LED's. That's why for 150 Led's you should set it as 50 pixels (or LED's), however when using effects with movement it really seems that each LED is controlled individually, but it isn't.
There are GRB and BRG types, I have both flavours.

brianmiller1956, 2021-01-16:
I suspected that it was actually doing something slightly unusual when the display moves.  An optical illusion!

Makuna, 2021-01-16:
When there is a chip that controls external LEDs (versus a single LED with the chip inside); then the count is the number of chips not the LEDs.  

As mentioned, the chip can control more than one RGB set of LEDs.  These sets for each chip are not individually addressable and all act together as they are electrically connected in series to the chip. I have seen diagrams with up to six sets of RGB LEDs controlled by a single chip but I have never seen one in real life.

brianmiller1956, 2021-01-16:
Interesting, it simply means that some patterns will maybe not look as good as single leds would, but it is a good way of using some “old” hardware and I am sure that there are many of these still in existence.

WerkAG, 2021-01-17:
Nowadays we can do a lot better with individually addressable led pixels, however in any case it all will depend on how programing in done. Despite being "old" they are quite capable led strings.
Video below was made in early Nov 2012 using LPD6803. Note: When video was shot, Lighting effects and Audio weren't fully in sync.

https://user-images.githubusercontent.com/14356332/104824903-181a4400-584e-11eb-966c-f1f0d82cb123.mp4

brianmiller1956, 2021-01-18:
Hi, I think I am making progress, I downloaded the WLED v11.1 zip file and loaded it into the Arduino ide, it successfully compiled for both the esp32 and esp8266, setups, I then duplicated the sketch as a second version, checked that it compiled then substituted your npbwrapper for the original, it would not compile, complaining that bool CanShow was not defined in npbwrapper  I compared yours to the original and found the CanShow routine, I copied it into your npbwrapper, the sketch now compiles, I have not uploaded either sketch to a controller yet but will do so in the next few days and see what happens, should be able to test some leds, hope this is ok

WerkAG, 2021-01-19:
Hi, I thing that my npbwrapper was for a WLED version older than v11.1. I haven't tried yet the new version.
Now that NeoPixelsBus includes support for the LPD6803, I believe WLED will add it to npbwrapper too, changes are quite simple. Its just a matter of waiting sometime, I think. :-)

brianmiller1956, 2021-01-19:
More progress!
I uploaded the unmodified v11.1 sketch to a wemos D1 mini clone, connected some ws2812 leds and it setup and ran fine.
I then uploaded the modified version with your npbwrapper modifications, the user interface was usable but the lpd6803 leds were not under any sort of control, I changed the data and clock pins to clock pin 0 and data pin 2 as per the original config, uploaded the code and the leds are working fine.
I will do some longer term testing to see how well they work.
Many thanks for your help.

Makuna, 2021-01-31:
2.6.1 released

closed 2021-01-31 by Makuna