Makuna/NeoPixelBus #389: Wrong times for WS2811? (NeoEsp32RmtSpeedWs2811)

CsharpLassi, 2020-12-22:
Hi,

i use a WS2811 strip with a wemos lolin32 lite board and the default values do not work for me:

`
const static DRAM_ATTR uint32_t RmtBit0 = Item32Val(300, 950); 
const static DRAM_ATTR uint32_t RmtBit1 = Item32Val(900, 350); 
const static DRAM_ATTR uint16_t RmtDurationReset = FromNs(300000); // 300us
`
What are these values because I found two data sheets with two different times:

[WS2811.pdf](https://github.com/Makuna/NeoPixelBus/files/5728373/WS2811.pdf)
[WS2811_V1_4.pdf](https://github.com/Makuna/NeoPixelBus/files/5728374/WS2811_V1_4.pdf)

The following settings work for me:

`
  const static DRAM_ATTR uint32_t RmtBit0 = Item32Val(500, 2000);
  const static DRAM_ATTR uint32_t RmtBit1 = Item32Val(1200, 1300);
  const static DRAM_ATTR uint16_t RmtDurationReset = FromNs(60000); // 60us
`
I would like to create a pull request, but I don't know where the old values come from.

Makuna, 2020-12-22:
I have an official spec sheet that is different; but similar except for one value.  T1L is supposed to be 200ns-420ns.  The ones you sent don't make sense as they generate a pulse width (total time of both and high and low) that varies dramatically from bit 0 to a bit 1.

I have tested this on official WS2811 chips and it works; but these chips come from China, the companies change things from batch to batch and clones are made that don't match specs.  Do you have a link to where you got them?

Did you try the values from the spec sheets you have to see if they work? I ask as the ones you list don't match any spec sheet for any chip I know of.  They are similar to first generation LEDs (big through hole LED with integrated chip).  These are the 400Kbps leds and there are methods for those specifically.  See Neo400KbpsMethod and NeoEsp32RmtSpeed400Kbps.

CsharpLassi, 2020-12-24:
Here the link:
https://de.aliexpress.com/item/2000165819.html
I bought them 2-3 years ago.

The methods NeoEsp32RmtSpeed400Kbps and Neo400KbpsMethod works not for me. Only the posted times work.
The times correspond to the first datasheet.

Makuna, 2020-12-24:
My spec sheets are from WorldSemi directly.  
[WS2811_--_201709_V1.1_17101018593714.pdf](https://github.com/Makuna/NeoPixelBus/files/5740217/WS2811_--_201709_V1.1_17101018593714.pdf)

Products from China may come with "clone" parts, so no, your strips may not have chips from WorldSemi or the batch you got they "changed" the specs.

+-50us is not achievable for the implementations of most the Arduinos supported by this library.

closed 2020-12-24 by Makuna