Makuna/NeoPixelBus #390: WS2801 on a fast CPU has data being sent to fast

Makuna, 2020-12-25:
**Describe the bug**
default SPI speed is 10Mhz when it can only handle 2Mhz
default bitbang on a fast CPU is faster than 2Mhz (esp8266 for example)

**Expected behavior**
Data rate must be kept at or below about 2Mhz.

**Minimal Sketch that reproduced the problem:**
Use the DotStar example.

Makuna, 2020-12-29:
I put it on the scope using an esp866 and I got the data rates. 
80mhz cpu, 330Khz data, 
160Mhz 1Mhz data

 so even at the highest speed of the cpu, it will never exceed the what the ws2801 should handle.

NOW, what I did spot was "ringing". Both the clock and the data pins when going low would ring causing spurious very short pulses after the real pulse of about 2ns. 

This means filtering is needed to keep the ringing down and you can see it in both software and hardware captures below.

The only other difference I spotted was that with hardware, the data line went low when the clock went low, 

[image]

while with software, the data would go low AFTER the clock went low.

[image]

Burn-In, 2020-12-29:
I have just test the different ESP8266SPI modes with 80 and 160MHz CPU speed, if it works or not.
Results:
ESP8266 CPU Frequency            80MHz   160MHz
NeoWs2801SpiMethod                 x              y
NeoWs2801Spi20MhzMethod      x              y
NeoWs2801Spi10MhzMethod      x              y
NeoWs2801Spi2MhzMethod        y              y 

Maybe it helps.

Makuna, 2021-01-25:
I can find nothing wrong with speeds so I am closing this.

closed 2021-01-25 by Makuna