Makuna/NeoPixelBus #401: APA106 occasionally displays bright white regardless of color provided

atanisoft, 2021-01-08:
**Describe the bug**
Intermittently APA106 can display white when sending `RgbColor(0)` (or any other color).

**Development environment (please complete the following information):**
 - OS: Linux
 - Build Environment CMake
 - Board target ESP32
 - Library version Latest

I've tested the adjusted timings for the RMT APA106 as:
```
    const static DRAM_ATTR uint32_t RmtBit0 = Item32Val(350, 1350);
    const static DRAM_ATTR uint32_t RmtBit1 = Item32Val(1350, 350);
```

So far it is working nicely with no ill effects.

Makuna, 2021-01-16:
https://github.com/Makuna/NeoPixelBus/pull/406
Platforms that have specific ability to fine tune the values have been adjusted

Makuna, 2021-01-31:
2.6.1 released

closed 2021-01-31 by Makuna