Makuna/NeoPixelBus #414: HtmlColor(RgbwColor color) constructor

coelner, 2021-01-26:
**Is your feature request related to a problem? Please describe.**
Yes, I like to generate quickly the html color code from a rgbw value. 

**Describe the solution you'd like**
a simple constructor which takes only the rgb from a rgbw value

**Describe alternatives you've considered**
```
HtmlColor(const RgbwColor& color)
    {
        Color = (uint32_t)color.R << 16 | (uint32_t)color.G << 8 | (uint32_t)color.B;
    }
```

**Additional context**
Nothing to add

Makuna, 2021-01-27:
I found a worthy additions that follows the tenants but makes it easier to do what you are trying.

```
HtmlColor html;
RgbwColor rgbw;
RgbColor rgb;

rgb = rgbw; // will remain invalid
html = rgbw; // will remain invalid
html = RgbColor(rgbw); // will be valid as it is explicit to Rgb and not automatic; and rgb is automatic
```
This is done by using the new "explicit" property for constructors in C++ 11.

Makuna, 2021-01-27:
https://github.com/Makuna/NeoPixelBus/pull/415

coelner, 2021-01-27:
Perfect, that seems to be good enough. Thanks

Makuna, 2021-01-31:
2.6.1 released

closed 2021-01-31 by Makuna