Makuna/NeoPixelBus #421: GBR color order support for LPD6803

altery, 2021-02-10:
Thank you for the new support of the LPD6803 controller! This allow me to migrate an old project to a new platform :) But the LED chain I use seems to expect the color data in a different order than the two that are already supported. I'd like to request the inclusion of

Lpd6803GbrFeature (-> green, blue red)

in Lpd6803ColorFeatures.

Makuna, 2021-02-10:
https://github.com/Makuna/NeoPixelBus/pull/423

altery, 2021-02-10:
Thanks! I'm sorry, I had to change the color order in this feature request, as it seems to actually RGB (instead of GBR) in my case - it seems the previous LED driver library had it implemented incorrectly and it was 'corrected' in client side code. Something I just got aware of after more in depth testing.

Makuna, 2021-02-11:
https://github.com/Makuna/NeoPixelBus/pull/425

altery, 2021-03-06:
Thanks for the quick implementation! I hope it's ok, if I resolve the issue myself?

closed 2021-03-06 by altery