Makuna/NeoPixelBus #422: Support for lower SPI frequencies

altery, 2021-02-10:
An old project I'm migrating to NeoPixelBus+ESP8266 uses a rather long SPI wire and does not work reliably with high bandwidth transfers rates. Would it be possible to extend the supported SPI frequencies with lower rates (such es 1MHz and 0.5MHz)?

Makuna, 2021-02-10:
I will add those.

You can provide your own speeds by using the following example for testing In the mean time...
```
class MySpiSpeed {
public:
    static const uint32_t Clock = 1000000L; // 1mhz;  2000000L = 2mhz
};

typedef Lpd6803MethodBase<TwoWireSpiImple<MySpiSpeed>> Lpd6803SpiMySpeedMethod;
```
Then you can use Lpd6803SpiMySpeedMethod for your custom speed.

Makuna, 2021-02-10:
https://github.com/Makuna/NeoPixelBus/pull/423

altery, 2021-03-06:
Thanks for the quick implementation! I hope it's ok, if I resolve the issue myself?

closed 2021-03-06 by altery

Makuna, 2021-03-06:
Not sure you have seen it, but I also added a dynamic SPI speed setting method.  Its talked about in the discussion but not yet exposed in the Wiki.

See https://github.com/Makuna/NeoPixelBus/discussions/433

altery, 2021-03-06:
No, had not seen it, thanks for pointing that out!