Makuna/NeoPixelBus #424: Lpd6803 data array is not initialized properly

altery, 2021-02-10:
**Describe the bug**
After initializing the library in Lpd6803 mode, the internal data array is initialized with all 0x00 bytes. When only setting a few pixels (for example for topology or color ordering tests), incorrect data (0x00 bytes) is fed to the Lpd6803 and the LEDs light up arbitrarily.

**To Reproduce**
```
const uint16_t PixelCount = 64;

NeoPixelBus<Lpd6803GrbFeature, Lpd6803Spi2MhzMethod> strip(PixelCount);

RgbColor red(128, 0, 0);
RgbColor green(0, 128, 0);
RgbColor blue(0, 0, 128);
RgbColor white(128);
RgbColor black(0);

strip.Begin();

strip.SetPixelColor(topo.Map(left, top), white);
strip.SetPixelColor(topo.Map(right, top), red);
strip.SetPixelColor(topo.Map(right, bottom), green);
strip.SetPixelColor(topo.Map(left, bottom), blue);

strip.Show();
```

**Expected behavior**
The not yet initialized pixel should probably be black.

The problems seems to stem from Lpd6803GenericMethod:

```
memset(_data, 0, _sizeData);
```

Lpd6803 expects every 2-byte data pixel to start with a set MSB, so every other byte should be 0x80 instead of 0x00.

As a workaround, all pixels in the strip can be initialized manually:

```
  for(int i = 0; i < PixelCount; i++) {
    strip.SetPixelColor(i, black);
  }
```

Makuna, 2021-02-11:
https://github.com/Makuna/NeoPixelBus/pull/425

Makuna, 2021-02-11:
Ok, found that it could be a pitfall to fall into with other methods/features, so instead the Begin() nows calls ClearTo(0).  The passed zero gets turned into a 'black' for any of the color types.

https://github.com/Makuna/NeoPixelBus/pull/426

altery, 2021-03-06:
Great, thanks!

closed 2021-03-06 by altery