Makuna/NeoPixelBus #429: Use MALLOC_CAP_DMA for DMA-based transfers

embedded-creations, 2021-02-13:
I noticed you're not using the `MALLOC_CAP_DMA` flag for buffers that will be used for DMA transfers to a peripheral. e.g.

https://github.com/Makuna/NeoPixelBus/blob/13b7d2e3776248785da3c08e581f1af9af86c8f5/src/internal/NeoEsp32I2sMethod.h#L231

From the Espressif Docs for SPI (there are similar requirements for I2S, and I don't know about RMT):

> If DMA is enabled for transfers, the buffers are required to be:
> Allocated in DMA-capable internal memory. If external PSRAM is enabled, this means using pvPortMallocCaps(size, MALLOC_CAP_DMA).
> ...
> If these requirements are not satisfied, the **transaction efficiency will be affected due to the allocation and copying of temporary buffers**.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#write-and-read-phases

There is a limited amount of DMA-capable memory, and it's not available in one contiguous block, so how you allocate it is important.  I need a lot of DMA-capable memory for SmartMatrix Library on the ESP32, you might want to search my repo for "MALLOC_CAP_DMA" to see how I use it as a reference. e.g.

https://github.com/pixelmatix/SmartMatrix/blob/9818a208566292f9b790a6c7ef80fc152a5526d3/src/MatrixEsp32Hub75Refresh_Impl.h#L171

Hope this helps!

Makuna, 2021-02-13:
Thanks!  This is relevant to the I2S, the RMT I don't believe so due my library using the "translate" feature where the actual buffers used by sending is managed by core.

I don't have any units with PSRAM to test this with though; so I need to rely on others for testing.

p.s.  Updating the title to be more specific.

embedded-creations, 2021-02-14:
If it makes any difference, I don't have any units with external PSRAM either (or at least have never plugged one in), and just made sure to use MALLOC_CAP_DMA in my code as I thought it was required for all devices.  Someone else added support for external PSRAM to my library and didn't have to change any DMA malloc code, it just worked.

Makuna, 2021-02-14:
Searched the actual IDF code, and you are correct.
https://github.com/espressif/esp-idf/blob/73db142403c6e5b763a0e1c07312200e9b622673/components/driver/i2s.c#L625
BOTH, the description table and the actual buffer data needed to be allocated with MALLOC_CAP_DMA defined.

Makuna, 2021-02-15:
https://github.com/Makuna/NeoPixelBus/pull/430

Makuna, 2021-03-21:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.6.2

closed 2021-03-21 by Makuna