Makuna/NeoPixelBus #445: ESP32 BitBang code is not in IRAM

s-hadinger, 2021-03-06:
**Describe the bug**
When compiling Tasmota for ESP32, we identified occasional timing issues in Bit Bang method (for a single WS2812 led), depending on code size. It appeared that timing was skewed because of Bit Bang code not in IRAM causing some occasional cache misses and loads from SPI Flash.

Diving deeper, it appears that latest version of GCC do not honor the `IRAM_ATTR` attribute for methods in templated classes. Hence the `send_pixels()` method is never moved to `IRAM` despite its declaration:

https://github.com/Makuna/NeoPixelBus/blob/e54a06736e8410e67c513c1da499026b6d9e24ec/src/internal/NeoEspBitBangMethod.h#L138-L142

The work-around applied in Tasmota is to explicitly create some method specializations with `IRAM_ATTR` applied:
https://github.com/arendst/Tasmota/pull/11241

Unfortunately this method is not sustainable and would require a big amount of code duplication. I haven't found any other fix or work-around. This may lead to finding an alternative to templates, or changing the linker script to force those methods in IRAM anyways. I don't have a simple solution.

**To Reproduce**
Compile and check map file. The method 

**Expected behavior**
Compile the sketch below, with the build option in `platform.ini`:
```
build_flags                 = -Wl,-Map,firmware.map
```

Then looking at the `firmware.map` file:
```
 .text._ZN17NeoEspBitBangBaseI17NeoEspSpeed800Mhz12NeoEspPinsetE11send_pixelsEPhS3_h
                0x00000000400d0ba4       0x66 .pio/build/m5stack-core-esp32/src/main.cpp.o
                0x00000000400d0ba4                _ZN17NeoEspBitBangBaseI17NeoEspSpeed800Mhz12NeoEspPinsetE11send_pixelsEPhS3_h
```

The `send_pixels` method is at address 0x0400d0ba4 which is not in the IRAM section, but in the Flash `.text` section

**Development environment (please complete the following information):**
 - OS: MacOS
 - Build Environment Platform.io 5.1.0
 - Board target m5stack-core-esp32 - espressif32 (Arduino ESP32)
 - Library version 2.6.1

**Minimal Sketch that reproduced the problem:**
```
#include <Arduino.h>
#include <NeoPixelBus.h>

NeoEspBitBangBase<NeoEspSpeed800Mhz, NeoEspPinset> strip;

void setup() {
}

void loop() {
  strip.send_pixels(nullptr, nullptr, 0);
}
```

Note: this sketch is not meant to be run, be just to check the compilation result and section of code.

**Additional context**

Makuna, 2021-03-06:
https://github.com/Makuna/NeoPixelBus/issues/321
I found a work around for the above related issue.  Maybe take a glance to see if that can be applied?  

I will take a look at some point; but it is not a priority as BitBang is not a suitable method for Esp8266 nor Esp32 due to low level interrupts that can not be blocked nor consistently controlled.

s-hadinger, 2021-03-07:
Thanks, following what's been done on other issues, I'm proposing a similar approach with #446

Makuna, 2021-03-13:
s-hadinger has made a pull that I have merged in.  This is a work around to the external issue.  This issue will remain OPEN to track reverting this pull when the compiler can correctly support attributes on template methods.

Makuna, 2021-03-21:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.6.2

closed 2021-03-22 by s-hadinger