Makuna/NeoPixelBus #450: Won't compile for ClearCore Arduino

tohox, 2021-03-19:
Hi,

Trying to compile the NeoPixelTest example for this particular board based on an ARM M4F, SAME53N19A chip and getting a bunch of missing declarations. Any hope of getting this to compile on this neat industrial Arduino variant?

Arduino: 1.8.13 (Windows 10), Board: "Teknic ClearCore"
In file included from ...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:114:0,
                 from ...\AppData\Local\Temp\arduino_modified_sketch_86674\NeoPixelTest.ino:15:
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h: In static member function 'static void NeoArmOtherSpeedBase<T_SPEEDPROPS>::send_pixels(uint8_t*, size_t, uint8_t)':
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:725:9: error: 'Pio' was not declared in this scope
         Pio* port;
         ^~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:725:14: error: 'port' was not declared in this scope
         Pio* port;
              ^~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:725:14: note: suggested alternative: 'Port'
         Pio* port;
              ^~~~
              Port
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:735:9: error: there are no arguments to 'pmc_set_writeprotect' that depend on a template parameter, so a declaration of 'pmc_set_writeprotect' must be available [-fpermissive]
         pmc_set_writeprotect(false);
         ^~~~~~~~~~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:735:9: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:736:9: error: there are no arguments to 'pmc_enable_periph_clk' that depend on a template parameter, so a declaration of 'pmc_enable_periph_clk' must be available [-fpermissive]
         pmc_enable_periph_clk((uint32_t)TC3_IRQn);
         ^~~~~~~~~~~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:739:13: error: 'TC_CMR_WAVE' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
             ^~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:739:13: note: suggested alternative: 'REG_TCC1_WAVE'
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
             ^~~~~~~~~~~
             REG_TCC1_WAVE
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:739:27: error: 'TC_CMR_WAVSEL_UP' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                           ^~~~~~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:739:27: note: suggested alternative: 'TC_CTRLBSET_LUPD'
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                           ^~~~~~~~~~~~~~~~
                           TC_CTRLBSET_LUPD
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:739:46: error: 'TC_CMR_TCCLKS_TIMER_CLOCK1' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:738:9: error: there are no arguments to 'TC_Configure' that depend on a template parameter, so a declaration of 'TC_Configure' must be available [-fpermissive]
         TC_Configure(TC1, 0,
         ^~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:740:9: error: there are no arguments to 'TC_Start' that depend on a template parameter, so a declaration of 'TC_Start' must be available [-fpermissive]
         TC_Start(TC1, 0);
         ^~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:742:19: error: 'g_APinDescription' was not declared in this scope
         pinMask = g_APinDescription[pin].ulPin; // Don't 'optimize' these into
                   ^~~~~~~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:742:19: note: suggested alternative: 'IcmDescriptor'
         pinMask = g_APinDescription[pin].ulPin; // Don't 'optimize' these into
                   ^~~~~~~~~~~~~~~~~
                   IcmDescriptor
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:746:28: error: 'union Tc' has no member named 'TC_CHANNEL'
         timeValue = &(TC1->TC_CHANNEL[0].TC_CV);  // the initial 'while'.
                            ^~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:747:28: error: 'union Tc' has no member named 'TC_CHANNEL'
         timeReset = &(TC1->TC_CHANNEL[0].TC_CCR);
                            ^~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:768:26: error: 'TC_CCR_CLKEN' was not declared in this scope
             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
                          ^~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:768:26: note: suggested alternative: 'TC_CTRLA_COPEN'
             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
                          ^~~~~~~~~~~~
                          TC_CTRLA_COPEN
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:768:41: error: 'TC_CCR_SWTRG' was not declared in this scope
             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
                                         ^~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:768:41: note: suggested alternative: 'TCC1_SWAP'
             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
                                         ^~~~~~~~~~~~
                                         TCC1_SWAP
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:788:9: error: there are no arguments to 'TC_Stop' that depend on a template parameter, so a declaration of 'TC_Stop' must be available [-fpermissive]
         TC_Stop(TC1, 0);
         ^~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h: In instantiation of 'static void NeoArmOtherSpeedBase<T_SPEEDPROPS>::send_pixels(uint8_t*, size_t, uint8_t) [with T_SPEEDPROPS = NeoArmOtherSpeedPropsWs2812x; uint8_t = unsigned char; size_t = unsigned int]':
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:84:29:   required from 'void NeoArmMethodBase<T_SPEED>::Update(bool) [with T_SPEED = NeoArmOtherSpeedBase<NeoArmOtherSpeedPropsWs2812x>]'
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:189:9:   required from 'void NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::Show(bool) [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoArmMethodBase<NeoArmOtherSpeedBase<NeoArmOtherSpeedPropsWs2812x> >]'
...\AppData\Local\Temp\arduino_modified_sketch_86674\NeoPixelTest.ino:77:16:   required from here
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:735:29: error: 'pmc_set_writeprotect' was not declared in this scope
         pmc_set_writeprotect(false);
         ~~~~~~~~~~~~~~~~~~~~^~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:736:30: error: 'pmc_enable_periph_clk' was not declared in this scope
         pmc_enable_periph_clk((uint32_t)TC3_IRQn);
         ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:739:72: error: 'TC_Configure' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                                                                        ^
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:740:17: error: 'TC_Start' was not declared in this scope
         TC_Start(TC1, 0);
         ~~~~~~~~^~~~~~~~
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:740:17: note: suggested alternative: 'dhcp_start'
         TC_Start(TC1, 0);
         ~~~~~~~~^~~~~~~~
         dhcp_start
...\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:788:16: error: 'TC_Stop' was not declared in this scope
         TC_Stop(TC1, 0);
         ~~~~~~~^~~~~~~~
exit status 1
Error compiling for board Teknic ClearCore.

Makuna, 2021-03-20:
What ever source you are using, it doesn't match the current code, the line numbers in the errors don't match up.

[This is line 725](https://github.com/Makuna/NeoPixelBus/blob/f16a39f7af5b8920cee969f2054516ee6f396238/src/internal/NeoArmMethod.h#L725)
 
From what I can read, it is falling through the various ARM implementation into the "other" which is assumed to be a Duo board; which yours is not.  Often this can happen for one of two reasons;
* the most common, the "ARDUINO_ARCH" symbol is not defined in the build environment you have (common with ARM). Some don't even try to follow the standard.  An example of a properly defined ARM symbol is "ARDUINO_ARCH_STM32F4" and example of a very poor one is "__MKL26Z64__".  Since you didn't include the compile line with the symbols that are defined, I couldn't see what your environment sets.  Usually a single line with a bunch of "-Dblah blah", where one is "-DARDUINO_ARCH_blah".
* the second most common, it is not directly supported as it doesn't match any of the currently supported ARM implementations enough to be covered.

tohox, 2021-03-20:
I checked the verbose compilation option and got the compiler arguments below. I believe the symbol you mention gets set to  "ARDUINO_ARCH_SAM".

Sorry for the wall of text that follows. Please let me know if I can somehow edit this to make it more palatable...

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_670955 -warnings=none -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_668569 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\THORSW~1\AppData\Local\Temp\arduino_modified_sketch_733989\NeoPixelTest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_670955 -warnings=none -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_668569 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\THORSW~1\AppData\Local\Temp\arduino_modified_sketch_733989\NeoPixelTest.ino
Using board 'clearcore' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Using core 'arduino' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Detecting libraries used...
"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -c -g -O3 -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -D__FPU_PRESENT=1 -DARDUINO=10813 -DARDUINO_ARM_ClearCore -DARDUINO_ARCH_SAM -D__ARM_FEATURE_DSP=1 -DCPPUTEST_USE_MEM_LEAK_DETECTION=0 -DUSB_VID=0x2890 -DUSB_PID=0x0022 "-DUSB_MANUFACTURER=\"Teknic, Inc.\"" "-DUSB_PRODUCT=\"Teknic ClearCore\"" -lc -DDEBUG -DUSBCON -DUSB_CONFIG_POWER=0 -D__CLEARCORE__ -D__SAME53N19A__ -DARM_MATH_CM4 "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino/api" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore/Third Party/SAME53/CMSIS/Device/Include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/libClearCore/inc" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/src/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/port/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore" "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_670955\\sk [... cut ...] rs\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore" "-IC:\\Users\\thorswill\\Documents\\Arduino\\libraries\\NeoPixelBus_by_Makuna\\src" "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_670955\\sketch\\NeoPixelTest.ino.cpp" -o "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_670955\\sketch\\NeoPixelTest.ino.cpp.o"`

Makuna, 2021-03-20:
Yeah, they tried to have a something defined but picked something so generic it is pretty useless.  ARM means almost nothing if all the ARMs don't expose a consistent HAL layer (and they don't).

So the key defines are ARDUINO_ARCH_SAM and __SAME53N19A__ with the latter being more important.

This chip is not supported in NeoPixelBus.
Is it supported in Adafruit's library?  

I wish SAM meant something common, like a consistent HAL layer that I could work on top of; but as it stands every chip has to have custom code to just provide a bitbang implementation let alone a proper hardware supported solution.  So it is very unlikely I would provide the support without an incentive from the company.

Feel free to look at the implementations that are present and see if they can work for you with changes.  The Nrf52x chips are also ARM and I support the PWM hardware for them as it nice easy interface to use but I thinks it is specific to that chipset.

NeoPixelBus\src\internal\NeoNrf52xMethod.h for the hardware PWM support.
NeoPixelBus\src\internal\NeoArmMethod.h for the various bitbang support.

tohox, 2021-03-21:
I understand perfectly well why you can't support every oddball chipset out there and thank you for looking into this.

I tried Adafruit's library as you suggested and while the code does compile and run without complaining I don't get any output from the LED pins I've tried or the timing must be off somehow.

Thanks again,

Makuna, 2021-03-21:
I scanned Adafruits library (we have some common code); I am not sure why you are not getting an error output when it compiles; as NONE of the symbols provided above are in their sets (arm or not).   Which should fall through to output the following compile error
```
Architecture not supported
```
If you can figure out what code of theirs is building into what ever sample you are using; then I might be able to look at "turning on" the same in my library.

tohox, 2021-03-22:
I added a bunch of #warning statements in Adafruit's NeoPixel library to see which parts were getting compiled and it seems it is only the global __AVR__ portion of NeoPixel.cpp that is getting accessed. None of the more specific areas are reached at compile time like __arm__, TEENSYDUINO, __MKL26Z64__, NRF52_SERIES, __SAMD21E17A__, __SAMD21E18A__ etc...

I know the ClearCore platform has a few deviations from the standard Arduino behavior. I'll review what the __ARM__ area of Adafruit's library does exactly and post back if useful to you.

Makuna, 2021-03-23:
Please give this little sketch a try and tell me what you see in the debug output.  This might lead to a general ARM bitbang that is more accurate than the current implementations.

```
#define DEMCR_TRCENA    0x01000000

#define DEMCR           (*((volatile uint32_t *)0xE000EDFC))
#define DWT_CTRL        (*(volatile uint32_t *)0xe0001000)
#define CYCCNTENA       (1<<0)
#define DWT_CYCCNT      ((volatile uint32_t *)0xE0001004)
#define CPU_CYCLES      *DWT_CYCCNT

static inline void cyclecount_reset(void)
{
    /* Enable DWT */
    DEMCR |= DEMCR_TRCENA; 
    *DWT_CYCCNT = 0;             
    /* Enable CPU cycle counter */
    DWT_CTRL |= CYCCNTENA;
}

void setup() {
    delay(5000); // some arms are temperamental with serial starting too fast 
    Serial.begin(115200);
    while (!Serial); // wait for serial attach
    
    cyclecount_reset();

    Serial.println(DWT_CTRL);
    Serial.println();
    Serial.println("Running...");
}

void loop() {
  uint32_t start = CPU_CYCLES;
  delay(2);
  uint32_t end = CPU_CYCLES;

  uint32_t delta = end - start;

  Serial.println(delta);
  delay(1000);
}
```

tohox, 2021-03-23:
Here you go:

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_971089 -warnings=default -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_96653 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\thorswill\Documents\Arduino\sketch_mar23a\sketch_mar23a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=0000_0000 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_971089 -warnings=default -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_96653 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\thorswill\Documents\Arduino\sketch_mar23a\sketch_mar23a.ino
Using board 'clearcore' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Using core 'arduino' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Detecting libraries used...
"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -c -g -O3 -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -D__FPU_PRESENT=1 -DARDUINO=10813 -DARDUINO_ARM_ClearCore -DARDUINO_ARCH_SAM -D__ARM_FEATURE_DSP=1 -DCPPUTEST_USE_MEM_LEAK_DETECTION=0 -DUSB_VID=0x2890 -DUSB_PID=0x0022 "-DUSB_MANUFACTURER=\"Teknic, Inc.\"" "-DUSB_PRODUCT=\"Teknic ClearCore\"" -lc -DDEBUG -DUSBCON -DUSB_CONFIG_POWER=0 -D__CLEARCORE__ -D__SAME53N19A__ -DARM_MATH_CM4 "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino/api" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore/Third Party/SAME53/CMSIS/Device/Include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/libClearCore/inc" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/src/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/port/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore" "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_971089\\sketch\\sketch_mar23a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
[... cut ...]
"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-objcopy" -O binary "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_971089/sketch_mar23a.ino.elf" "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_971089/sketch_mar23a.ino.bin"
"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-size" -A "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_971089/sketch_mar23a.ino.elf"
Sketch uses 132632 bytes (26%) of program storage space. Maximum is 507904 bytes.
Global variables use 8684 bytes of dynamic memory.`

Makuna, 2021-03-23:
I mean, upload and capture the serial output ;-)

tohox, 2021-03-23:
Doh! Not sure how many iteration you need but here are a few...

19:55:47.233 -> 1073741825
19:55:47.233 -> 
19:55:47.233 -> Running...
19:55:47.233 -> 240124
19:55:48.260 -> 240147
19:55:49.249 -> 240147
19:55:50.278 -> 240146
19:55:51.256 -> 240149
19:55:52.244 -> 240149
19:55:53.274 -> 240147
19:55:54.255 -> 240147
19:55:55.283 -> 240146
19:55:56.270 -> 240150
19:55:57.293 -> 240146
19:55:58.274 -> 240144
19:55:59.297 -> 240147
19:56:00.275 -> 240147
19:56:01.302 -> 240145
19:56:02.282 -> 240143
19:56:03.265 -> 240153
19:56:04.293 -> 240146
19:56:05.273 -> 240145
19:56:06.297 -> 240152
19:56:07.278 -> 240146
19:56:08.317 -> 240150
19:56:09.289 -> 240142
19:56:10.314 -> 240150
19:56:11.289 -> 240148
19:56:12.313 -> 240153
19:56:13.292 -> 240153
19:56:14.313 -> 240151
19:56:15.288 -> 240145
19:56:16.315 -> 240148
19:56:17.294 -> 240148
19:56:18.324 -> 240148
19:56:19.302 -> 240150
19:56:20.330 -> 240150
19:56:21.308 -> 240148
19:56:22.332 -> 240145

Makuna, 2021-03-24:
Well that is interesting.  The ESP32 Bitbang method could be modified to provide a reasonable version for many ARM processors that support this special cycle counter.  A lot of them don't (M0s for example) but at least in this case yours does; otherwise the numbers would have been zero like it does on the only ARM I own (a M0).

Stay tuned.

Makuna, 2021-03-24:
Something more to try quickly.  Take a NeoPixelBus sample (or your favorite sketch using NeoPixelBus).

At the top where you include NeoPixelBus, make sure it looks like this ....
```
#define __MK20DX128__
#include <NeoPixelBus.h>
#undef __MK20DX128__
```
And see if that works.
These ARMs already uses the cycle count model (shared from Adafruits library).

tohox, 2021-03-24:
So using your NeoPixelTest.ino sketch modified as so:

    #include "ClearCore.h"
    #define __MK20DX128__
    #include <NeoPixelBus.h>
    #undef __MK20DX128__  
    const uint16_t PixelCount = 15;
    const uint8_t PixelPin = M2_INB;  
    #define colorSaturation 128
    NeoPixelBus<NeoRgbwFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);

I get the following when compiling:

`
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=2890_8022 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_57881 -warnings=none -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_700066 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\THORSW~1\AppData\Local\Temp\arduino_modified_sketch_533984\NeoPixelTest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=2890_8022 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_57881 -warnings=none -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_700066 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\THORSW~1\AppData\Local\Temp\arduino_modified_sketch_533984\NeoPixelTest.ino
Using board 'clearcore' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Using core 'arduino' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Detecting libraries used...
"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -c -g -O3 -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -D__FPU_PRESENT=1 -DARDUINO=10813 -DARDUINO_ARM_ClearCore -DARDUINO_ARCH_SAM -D__ARM_FEATURE_DSP=1 -DCPPUTEST_USE_MEM_LEAK_DETECTION=0 -DUSB_VID=0x2890 -DUSB_PID=0x0022 "-DUSB_MANUFACTURER=\"Teknic, Inc.\"" "-DUSB_PRODUCT=\"Teknic ClearCore\"" -lc -DDEBUG -DUSBCON -DUSB_CONFIG_POWER=0 -D__CLEARCORE__ -D__SAME53N19A__ -DARM_MATH_CM4 "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino/api" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore/Third Party/SAME53/CMSIS/Device/Include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/libClearCore/inc" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/src/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/port/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\ [... cut ...]
                                 register
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoArmMethod.h:176:50: error: 'portClearRegister' was not declared in this scope
         volatile uint8_t* clr = portClearRegister(pin);
                                 ~~~~~~~~~~~~~~~~~^~~~~
Using library NeoPixelBus_by_Makuna at version 2.6.1 in folder: C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus_by_Makuna 
exit status 1
Error compiling for board Teknic ClearCore.
`

Makuna, 2021-03-24:
Thanks, just more proof of the fragmentation of ARM implementations for Arduino.  You will have to wait for my new ARM bitbang method I am working on.

Makuna, 2021-03-24:
Ok, I just published a [new branch here](https://github.com/Makuna/NeoPixelBus/tree/ArmBitBang) that contains the general purpose ARM bit bang methods.  It is currently only enabled for your specific board setup.

So, fetch this branch to your machine (you can put it directly over the installed version in the documents\arduino\libraries\NeoPixelBus* ) and give it a try.

I would suggest starting with the `NeoArmBitBangWs2812xMethod`.

tohox, 2021-03-25:
I can see NeoArmBitBangMethod.cpp but not NeoArmBitBangWs2812xMethod.cpp. Is that the one you meant?

Unfortunately they both generate a `'xyz' was not declared in this scope` when used to initialize a strip object.

Makuna, 2021-03-25:
The "method" is the second part of the NeoPixelBus template construction. Instead of Neo800KbpsMethod use NeoArmBitBangWs2812xMethod like...
```
NeoPixelBus<NeoRgbwFeature, NeoArmBitBangWs2812xMethod> strip(PixelCount, PixelPin);
```
No need for any special defines (so remove ones I have asked you to try).

If this gives you a build error, then more than likely you did not clone the branch into the correct location.

tohox, 2021-03-25:
Yes, this was my understanding as well.

I copied the ArmBitBang branch over the existing C:\Users<my user name>\Documents\Arduino\libraries\NeoPixelBus by Makuna\ and restarted the IDE in case that was required to take the changed files into account. I also verified the time stamps in the src\internal directory and the files were indeed overwritten. I can see the NeoArmBitBangMethod.h  file in there but the compiler apparently doesn't... I erase the complete folder instead of simply overwriting it and post back.

My current code is as follows:

    #include "ClearCore.h"
    #include <NeoPixelBus.h>
    const uint16_t PixelCount = 15;
    const uint8_t PixelPin = M2_INB;
    #define colorSaturation 128
    NeoPixelBus<NeoRgbwFeature, NeoArmBitBangMethod> strip(PixelCount, PixelPin);
    void setup()
    {
        Serial.begin(115200);
        while (!Serial); // wait for serial attach
        strip.Begin();
        strip.Show();
    }
    #I commented out every in loop for now....

tohox, 2021-03-25:
So I started anew with the same result unfortunately...

Here goes:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=2890_8022 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_788692 -warnings=none -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_256622 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\thorswill\Documents\Arduino\NeoPixelTest\NeoPixelTest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\thorswill\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\thorswill\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\thorswill\Documents\Arduino\libraries -fqbn=ClearCore:sam:clearcore -vid-pid=2890_8022 -ide-version=10813 -build-path C:\Users\THORSW~1\AppData\Local\Temp\arduino_build_788692 -warnings=none -build-cache C:\Users\THORSW~1\AppData\Local\Temp\arduino_cache_256622 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.CMSIS.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\thorswill\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\thorswill\Documents\Arduino\NeoPixelTest\NeoPixelTest.ino
Using board 'clearcore' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Using core 'arduino' from platform in folder: C:\Users\thorswill\AppData\Local\Arduino15\packages\ClearCore\hardware\sam\1.0.14
Detecting libraries used...
"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -c -g -O3 -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=120000000L -D__FPU_PRESENT=1 -DARDUINO=10813 -DARDUINO_ARM_ClearCore -DARDUINO_ARCH_SAM -D__ARM_FEATURE_DSP=1 -DCPPUTEST_USE_MEM_LEAK_DETECTION=0 -DUSB_VID=0x2890 -DUSB_PID=0x0022 "-DUSB_MANUFACTURER=\"Teknic, Inc.\"" "-DUSB_PRODUCT=\"Teknic ClearCore\"" -lc -DDEBUG -DUSBCON -DUSB_CONFIG_POWER=0 -D__CLEARCORE__ -D__SAME53N19A__ -DARM_MATH_CM4 "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino/api" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore/Third Party/SAME53/CMSIS/Device/Include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/libClearCore/inc" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/src/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/port/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore" "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_788692\\sketch\\NeoPixelTest.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for NeoPixelBus.h: [NeoPixelBus-ArmBitBang@2.6.2]
ResolveLibrary(NeoPixelBus.h)
  -> candidates: [NeoPixelBus-ArmBitBang@2.6.2]
[... cut ...]
C:\Users\thorswill\Documents\Arduino\NeoPixelTest\NeoPixelTest.ino: In function 'void setup()':
NeoPixelTest:26:11: error: request for member 'Begin' in 'strip', which is of non-class type 'int'
     strip.Begin();
           ^~~~~
NeoPixelTest:27:11: error: request for member 'Show' in 'strip', which is of non-class type 'int'
     strip.Show();
           ^~~~
Using library NeoPixelBus-ArmBitBang at version 2.6.2 in folder: C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang 
exit status 1
'NeoArmBitBangMethod' was not declared in this scope

Makuna, 2021-03-26:
Please try to minimize the amount of copied compile output.  Only need the errors at the end and one line above them that includes the -DARDUINO_ARH_ARM part.

Look at your example, are you using the `NeoArmBitBangWs2812xMethod` I asked you to use?  
This is what you have
```
NeoPixelBus<NeoRgbwFeature, NeoArmBitBangMethod> strip(PixelCount, PixelPin);
```
This is what I asked you to use
```
NeoPixelBus<NeoRgbwFeature, NeoArmBitBangWs2812xMethod> strip(PixelCount, PixelPin);
```
Do you see the difference?

tohox, 2021-03-27:
Hi,

Yes, I first tried with NeoArmBitBangWs2812xMethod and since the compiler reported it as undeclared I then tried NeoArmBitBangMethod with the same result.

I will gladly reduce the amount of clutter in any subsequent posts.

Makuna, 2021-03-27:
I made a small change, maybe it will help clarify any build issues.

At the top of your sketch, before you include NeoPixelBus.h, add
```
#define USE_ARM_DWT_BITBANG
#include <NeoPixelBus.h>
```

This will require that you grab the latest version of custom branch (https://github.com/Makuna/NeoPixelBus/tree/ArmBitBang)

I would suggest you remove your current NeoPixelBus library from the library manager and make sure its gone in Arduino\libraries\ folder.  Then re-clone/download this ArmBitBang branch into your library folder.

tohox, 2021-03-30:
Sorry for the delayed response.

Completely removed library and copied latest ArmBitBang branch. Same result unfortunately. NeoArmBitBangWs2812xMethod still undeclared (see errors below). I then tried deleting the ArmBitBang library folder before recompiling just to make sure there weren't some nested leftover copies somewhere interfering and I immediately got a missing NeoPixelBus.h error which seems to indicate it was indeed using the right library folder and that there weren't any others interfering.

`"C:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -c -g -O3 -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=120000000L -D__FPU_PRESENT=1 -DARDUINO=10813 -DARDUINO_ARM_ClearCore -DARDUINO_ARCH_SAM -D__ARM_FEATURE_DSP=1 -DCPPUTEST_USE_MEM_LEAK_DETECTION=0 -DUSB_VID=0x2890 -DUSB_PID=0x0022 "-DUSB_MANUFACTURER=\"Teknic, Inc.\"" "-DUSB_PRODUCT=\"Teknic ClearCore\"" -lc -DDEBUG -DUSBCON -DUSB_CONFIG_POWER=0 -D__CLEARCORE__ -D__SAME53N19A__ -DARM_MATH_CM4 "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino/api" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore/Third Party/SAME53/CMSIS/Device/Include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/libClearCore/inc" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/src/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14/Teknic/LwIP/LwIP/port/include" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\cores\\arduino" "-IC:\\Users\\thorswill\\AppData\\Local\\Arduino15\\packages\\ClearCore\\hardware\\sam\\1.0.14\\variants\\clearcore" "-IC:\\Users\\thorswill\\Documents\\Arduino\\libraries\\NeoPixelBus-ArmBitBang\\src" "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_971853\\sketch\\NeoPixelTest.ino.cpp" -o "C:\\Users\\THORSW~1\\AppData\\Local\\Temp\\arduino_build_971853\\sketch\\NeoPixelTest.ino.cpp.o"
In file included from C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/NeoPixelBus.h:119:0,
                 from C:\Users\thorswill\Documents\Arduino\NeoPixelTest\NeoPixelTest.ino:2:
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h: In static member function 'static void NeoArmOtherSpeedBase<T_SPEEDPROPS>::send_pixels(uint8_t*, size_t, uint8_t)':
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:765:9: error: 'Pio' was not declared in this scope
         Pio* port;
         ^~~
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:765:14: error: 'port' was not declared in this scope
         Pio* port;
              ^~~~
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:765:14: note: suggested alternative: 'Port'
         Pio* port;
              ^~~~
              Port
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:775:9: error: there are no arguments to 'pmc_set_writeprotect' that depend on a template parameter, so a declaration of 'pmc_set_writeprotect' must be available [-fpermissive]
         pmc_set_writeprotect(false);
         ^~~~~~~~~~~~~~~~~~~~
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:775:9: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:776:9: error: there are no arguments to 'pmc_enable_periph_clk' that depend on a template parameter, so a declaration of 'pmc_enable_periph_clk' must be available [-fpermissive]
         pmc_enable_periph_clk((uint32_t)TC3_IRQn);
         ^~~~~~~~~~~~~~~~~~~~~
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:779:13: error: 'TC_CMR_WAVE' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
             ^~~~~~~~~~~
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:779:13: note: suggested alternative: 'REG_TCC1_WAVE'
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
             ^~~~~~~~~~~
             REG_TCC1_WAVE
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:779:27: error: 'TC_CMR_WAVSEL_UP' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                           ^~~~~~~~~~~~~~~~
C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang\src/internal/NeoArmMethod.h:779:27: note: suggested alternative: 'TC_CTRLBSET_LUPD'
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
                           ^~~~~~~~~~~~~~~~
[... cut ...]
Using library NeoPixelBus-ArmBitBang at version 2.6.2 in folder: C:\Users\thorswill\Documents\Arduino\libraries\NeoPixelBus-ArmBitBang 
exit status 1
'NeoArmBitBangWs2812xMethod' was not declared in this scope
`

Installing the [Clearcore wrapper](https://www.teknic.com/files/downloads/ClearCore_Arduino_Wrapper_Installer.zip) from your end might prove simpler to debug. Even though you don't have the hardware you could still see if it compiles or not. Once you've got it compiling on your end I can validate if it works on the hardware.

Thanks for your patience!

Makuna, 2021-03-30:
If Clearcore team wants to support Arduino correctly (with a board manager install) I would give it try.  I am over the "install Arduino support from our site only" crap; as almost all of them have some sort of install/build problems or even standard Arduino support problems; let alone a poor path to upgrading the versions.  

I just switched from ARDUINO_ARCH_ARM to ARDUINO_ARCH_SAM; for now.  The ARDUINO_ARCH_SAM is specific to your build environment, all mine use ARDUINO_ARCH_ARM.