Makuna/NeoPixelBus #451: ESP32/I2S/SK6812: max. LED module count 509 - if num of LED modules >= 510 LEDs light up erratically

mirko, 2021-03-20:
**Setup**

LED modules: SK6812 RGBW LEDs
Count LED modules: >= 510 LEDs
Platform: ESP32
Addressing: I2S

**Describe the bug**
If (defined) number of LED modules exceeds 509 (-> >= 510), addressing is erratic. Only parts of strip is being addressed (usually the first number of LEDs) in a different color than selected.
It kind of feels like an overflow somewhere, which corrupts the data, hence LEDs seem to light up erratically.

**Expected behavior**
>= 510 LED modules addressable the same way as <510 LED modules.

Makuna, 2021-03-20:
Did you try to use the RMT method and see if it has the same problem?
Did you try a very minimal sketch that reproduces the problem?  

a minimal sketch usually leaves around 312K of ram available.  

510 x 4 = 2040 bytes for the pixel buffer (pixels x rgbw)
2040 x 4 = 8150 bytes for the i2s send buffer (send buffer translation for i2s)

Memory shouldn't be an issue as long as your sketch isn't allocating a lot.

mirko, 2021-03-20:
I don't have a minimal sketch or sketch at all, but totally understand if that's necessary in order to pursue this issue.
Firmware is esphome (with version 2.6.1 (vanilla) of neopixel, though) and for the time being I don't use anything else among driving the LED strip.
The 509 limit stays the same, no matter what I do (even if I allocate a couple of KB beforehand (made sure they're not optimised away)).
For what it's worth: with an ESP8266 (esphome as well) using UART/DMA everything works as expected.

Makuna, 2021-03-21:
2.5.3!  There might be your problem.  The ESP32 I2s implementation was "not recommended" due to changes in the Esp32 Arduino core for over a year, maybe two. The default was changed to RMT methods.  

I didn't get it fixed until February of this year; and it seems I have not made a NEW release that includes these changes.  Its just in the active GITHUB repro.

This is what a minimal sketch would look like ...
```
#include <NeoPixelBus.h>

#define BUS_PIN 4
#define BUS_COUNT 510

NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method> strip(BUS_COUNT,BUS_PIN); // using i2s
// NeoPixelBus<NeoGrbwFeature, NeoEsp32Rmt3Sk6812Method> strip(BUS_COUNT,BUS_PIN); // using RMT

void setup() {
  strip.Begin();
  strip.Show();

  // set the last pixel to red
  strip.SetPixelColor(BUS_COUNT - 1, RgbwColor(128,0,0,0));
}

void loop() {
  strip.RotateRight(1);
  strip.Show();
  delay(66);
}
```

mirko, 2021-03-21:
> 2.5.3! There might be your problem.

As mentioned I used 2.6.1 (while the esphome fork is based on 2.5.3) but I'm happy to try 2.6.2 as soon as available and provide feedback - thanks!

mirko, 2021-03-21:
While not yet published, tried with master (=2.6.2) and indeed the rratic behaviour is gone.
*However* the maximum amount of LEDs I can address is still 509.
While specifying >509 LEDs doesn't cause the first 509 to behave weird anymore, it's (still) the maximum amount I can drive.

I changed it to a lower number and indeed the change takes place, ensuring the changes actually make it into the firmware.

Makuna, 2021-03-22:
Using the following sketch, it correctly sends all the data.  The only thing I see in the logic analyzer is that the i2s hardware is sending the stream twice with the correct RESET time between them which should not effect proper SK6812.  Maybe your LEDs are not following the spec for SK6812.  Try the WS2812x method (NeoEsp32I2s1Ws2812xMethod) and also, try the RMT method (NeoEsp32Rmt3Sk6812Method).

```
#include <NeoPixelBus.h>

#define BUS_PIN 4
#define BUS_COUNT 512

NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method> strip(BUS_COUNT,BUS_PIN); // using i2s
// NeoPixelBus<NeoGrbwFeature, NeoEsp32Rmt3Sk6812Method> strip(BUS_COUNT,BUS_PIN); // using RMT

void setup() {
  strip.Begin();
  strip.Show();
  
  // set the last pixel to red
  strip.SetPixelColor(BUS_COUNT - 3, RgbwColor(255,0,0,0));
  strip.SetPixelColor(BUS_COUNT - 4, RgbwColor(0,255,0,0));
  strip.SetPixelColor(BUS_COUNT - 2, RgbwColor(0,0,255,0));
  strip.SetPixelColor(BUS_COUNT - 1, RgbwColor(0,0,0,255));
}

void loop() {
  strip.Dirty();
  strip.Show();
  delay(500);
}
```

Makuna, 2021-03-23:
great, closing this issue then.

closed 2021-03-23 by Makuna