Makuna/NeoPixelBus #457: Error after update to 2.6.2 - ESP32

bfaliszek, 2021-03-31:
**Describe the bug**
After update from 2.6.1 to 2.6.2, the ESP32 looping below error

**Expected behavior**
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40088e58
file: "/Users/bfaliszek/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp32I2sMethod.h" line 227
func: void NeoEsp32I2sMethodBase<T_SPEED, T_BUS, T_INVERT>::construct(uint16_t, size_t, size_t) [with T_SPEED = NeoEsp32I2sSpeedWs2812x; T_BUS = NeoEsp32I2sBusOne; T_INVERT = NeoEsp32I2sNotInverted; uint16_t = short unsigned int; size_t = unsigned int]
expression: pixelCount >= 2 ? ESP_OK : ESP_ERR_INVALID_ARG

ELF file SHA256: 0000000000000000

Backtrace: 0x4008899f:0x3ffe3b30 0x40088e5b:0x3ffe3b50 0x400e7b85:0x3ffe3b70 0x401a1683:0x3ffe3b90 0x40084e0d:0x3ffe3bb0 0x40084f43:0x3ffe3c00 0x4007925f:0x3ffe3c40 0x400792c5:0x3ffe3c70 0x400792d0:0x3ffe3ca0 0x4007947d:0x3ffe3cc0 0x400806e6:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
```

**Development environment (please complete the following information):**
 - OS: macOS 11.3
 - Build Environment: Arduino IDE v.1.8.13
 - Board target: ESP32
 - Library version: v2.6.2

Makuna, 2021-03-31:
The key thing here is that you have a bus pixel count of 1 correct?

There is a problem with the ESP32 i2s in that the DMA hardware unit is too fast to support a single pixel and maintain consistency.  Just increase the pixel count to two and ignore the second pixel completely and it will work fine.

I am leaving this open as I will consider changing the code to just increment the count to two if you supply only one; but I was concerned that user make make assumptions about the count rather than use the `strip.GetPixelCount()` and I hadn't worked through all the scenarios yet to understand would any cause problems.

bfaliszek, 2021-04-01:
Yes, I have 1 pixel.
Everything works fine under 2.6.1.
```NeoPixelBrightnessBus <NeoGrbFeature, Neo800KbpsMethod> Neo_Strip (NeoPixel_NUMPIXELS, NeoPixel_DATA_PIN);```
When I change NeoPixel_NUMPIXELS to 2 or 3 it's also ok.

Under 2.6.2 I get the same error as in the beginning, even if I set NeoPixel_NUMPIXELS to 2 or 3.

Makuna, 2021-04-01:
With the latest builds, if you set the pixel count to 2 or greater, YOU WILL NOT GET THAT SAME ERROR MESSAGE.  If you are, then you did not flash your ESP32 correctly.  That message is only generated if the count is below 2, its even in the text above `pixelCount >= 2 ? ESP_OK : ESP_ERR_INVALID_ARG`.

PR #459 (2021-04-05): ESP32 i2s: instead of runtime exception, silently upgrade pixelCount
Fixes #457

This workaround, compared to the old implementation of throwing an exception,
is simpler, makes the compiled program 260 smaller, and unbreaks existing code
that uses a pixel count of 1 that used to work prior to 2cce77e9 (v2.6.2).

This fix basically does the same thing as the recommended fix on the caller
side, except without bothering the caller. It does make `.GetPixelCount()`
return 2 instead of the provided number of pixels, but the resulting program is
arguably less broken than it is with the failing check  [... cut]
commit cc17ec6f (2021-04-05): ESP32 i2s: instead of runtime exception, silently upgrade pixelCount (#459)

Fixes #457

This workaround, compared to the old implementation of throwing an exception,
is simpler, makes the compiled program 260 smaller, and unbreaks existing code
that uses a pixel count of 1 that used to work prior to 2cce77e9 (v2.6.2).

This fix basically does the same thing as the recommended fix on the caller
side, except without bothering the caller. It does make `.GetPixelCount()`
return 2 instead of the provided number of pixels, but the resulting program is
arguably less broken than it is with the failin [... cut]
closed 2021-04-05 by Makuna