Makuna/NeoPixelBus #458: Support for Ws2812-C

SexyRoyalBlue, 2021-04-03:
The timings on the C version of the LEDs is sadly diffrent. It wouldn't matter so much if the reset timing wasn't so much bigger. I would like to see other types supported too if that is possible

Code 	WS2812B 	     WS2812C
T0H 	250ns~550ns    220ns~380ns
T1H 	650ns~950ns    580ns~1µs
T0L 	700ns~1µs 	     580ns~1µs
T1L 	300ns~600ns    220ns~420ns
reset 	>50µs 	             >280µs

Above are the diffrent timings and as you can see the reset timing is over five times larger which makes for a great rainbow flicker show after a few leds

Makuna, 2021-04-03:
Did you try the `NeoWs2812xMethod`?  It has the longer reset time already.

What platform/Arduino are you using?

SexyRoyalBlue, 2021-04-04:
I don't use the lib myself but instead use wled and I'm using an esp 8266

Makuna, 2021-04-04:
I don't understand.  You are requesting support for something you don't use?  

I don't have every LED type to test with, I have to rely on "users" often to do testing and I am willing to work with them to add the support.  Which means you need to know which method you do use and be able to try methods to confirm they work or not.

The Spec actually states it differently than you did.

数据传输时间
T0H 0 码， 高电平时间 220ns ~ 380ns
T1H 1 码， 高电平时间 580ns ~ 1µs
T0L 0 码， 低电平时间 580ns ~ 1µs
T1L 1 码， 低电平时间 580ns ~ 1µs
RES 帧单位，低电平时间 280µs 以上

 translates roughly to 

Data transfer time
T0H 0 measurement, high-level time 220ns to 380ns
T1H 1 measurement, high-level time 580ns to 1 us
T0L 0 measurement, low time 580ns to 1 us
T1L 1 measurement, low-level time 580ns to 1 us
RES frame unit, low-level time of more than 280μs

These are very flexible definitions pretty much makes them compatible and very easy to support.

**These are within the current ranges of what the library already does with WS2812x method.**

closed 2021-04-04 by Makuna