Makuna/NeoPixelBus #475: Publish to Conan Center

mcandre, 2021-06-05:
**NOTE:  If you are seeking help or have questions, this is NOT the place to do it.**  For questions and support, jump on Gitter and ask away.  
[image]

**Is your feature request related to a problem? Please describe.**

Publishing this library to ConanCenter will simplify the installation process for many users.

**Describe the solution you'd like**

Publish NeoPixelBus to Conan Center.

https://conan.io/center/

Makuna, 2021-06-05:
This is an Arduino library, the installation is already part of the Arduino Library Manager installation process.  Why would this make it any simplier?

Makuna, 2022-06-05:
No response to question for over a year; closing.

closed 2022-06-05 by Makuna