Makuna/NeoPixelBus #476: Execption 0 with ESP8266 v3.0.0

christianBricaud, 2021-06-06:
**Describe the bug**
With version V3.0.0 of ESP8266, we get an exception 0 (illegal instruction) if we use EEPROM.begin () or LITTLEFs.begin () after using strip.begin().

**To Reproduce**
Use your NeoPixelTest.ino and add EEPROM.begin or LITTLEFS.begin() after strip.begin()

**Development environment 
 - OS:  Win10
 - Build Environment : Arduino IDE v.1.8.15
 - Board target : LOLIN (WEMOS) D1 R2 mini
 - Library version v2.3.6.

**Minimal Sketch that reproduced the problem:**
#include <NeoPixelBus.h>
#include <EEPROM.h>
const uint16_t PixelCount = 4; // this example assumes 4 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 2;  // make sure to set this to the correct pin, ignored for Esp8266
NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
void setup()
{
    Serial.begin(115200);
    while (!Serial); // wait for serial attach

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();
    EEPROM.begin(512);
    Serial.println();
    Serial.println("Running...");
}

void loop()
{
}

Result 👍 Exception 0: Illegal instruction
PC: 0x40201034: NeoEsp8266DmaMethodBase ::i2s_slc_isr() at C:\Users\..\NeoPixelBus_by_Makuna\src/internal/NeoEsp8266DmaMethod.h line 458
EXCVADDR: 0x00000000

Makuna, 2021-06-06:
Are you sure you have the library version correct?  That is old version you entered.

Makuna, 2021-06-06:
Mmm, it seems that i2s_slc_isr is being placed into IROM and not IRAM as requested by the use of IRAM_ATTR attribute.  This is something new with v3 I suspect.

Makuna, 2021-06-07:
Yup, update to the newer GCC causes this active issue with the compiler where it ignores attributes on static methods in template classes.
Looking at applying a workaround.

Makuna, 2021-06-11:
https://github.com/Makuna/NeoPixelBus/pull/482

@christianBricaud Could you give this a try to confirm it works for you.

christianBricaud, 2021-06-13:
Tested for a dozen hours on my application (with release 3.0.0.) without any problem. Thank you for your responsiveness and well done.

Makuna, 2021-06-13:
v2.6.4

closed 2021-06-13 by Makuna