Makuna/NeoPixelBus #477: Support for ESP32C3

s-hadinger, 2021-06-06:
**Is your feature request related to a problem? Please describe.**
Are there short term plans to support ESP32C3?

**Describe the solution you'd like**
ESP32C3 has more or less the same features as ESP32, but reduces number of GPIOs and channels (RMT, I2S, no DAC...)

**Describe alternatives you've considered**
Currently compilation fails on ESP32C3

**Additional context**
We're currently porting Tasmota to ESP32C3, and NeoPixelBus is a major part of it. Also happy to help if needed.

Makuna, 2021-06-06:
I have been waiting until an official Arduino support release.  Supporting C2 was painful as I had to keep switching back and forth to work on ESP32 and then uninstall and reinstall C2 to work on it.  If it was less painful I would have jumped on it already.

I still don't even see C2 in the released version.  Is there something I am missing?

s-hadinger, 2021-06-07:
We have started working on C3 in Tasmota. The toolchain is ready, and Arduino framework has "Alpha preview with support for ESP32S2 and ESP32C3" in 2.0.0-alpha1, using esp-idf 4.4.

It's still early stage, but enough to compile Tasmota and make it boot on a prototype dev board.

The NeoPixelBus compilation fails at least for two reasons:
- there was some headers locations changes in 4.4 for C3. This is painful but hopefully not so hard to fix
- the feature set of C3 is different from ESP32: for ex less I2S buses, less RMT channels...

Jason2866, 2021-06-07:
@Makuna  If you use Platformio installing the needed framework and toolchain is just copy & paste.
I have uploaded all necessary toolchain to Platformio registry. Our Arduino framework ESP32-S2 / C3 is based on latest IDF4.4 git and latest Arduino ESP32 git version.
This Platformio environment entry builds the Tasmota ESP32-C3 variant
https://github.com/arendst/Tasmota/blob/311e7ad661fed130d8b6d5a4b042484facbd9dac/platformio_tasmota_cenv_sample.ini#L54-L74
`platformio run -e tasmota32c3`

Makuna, 2021-06-07:
@s-hadinger I created a branch with the changes needed if you please confirm it, I will then merge it into the primary branch.

https://github.com/Makuna/NeoPixelBus/tree/Esp32C3

Jason2866, 2021-06-08:
@s-hadinger  jumped in. @Makuna wow that was fast. Thx!
By using the branch and compiling for the C3 
[this](https://github.com/Makuna/NeoPixelBus/files/6614684/Error_log.txt) compile errors occour

Makuna, 2021-06-08:
@Jason2866 Thanks, missed a few spots its seems.  
I have just updated that branch again, give it another try.  The definitions of HSPI_HOST and VSPI_HOST are no longer supported in C3.  Also missed defaults for C3 is not I2S methods.

Jason2866, 2021-06-08:
@Makuna Great progress!
It looks like just a include is wrong
```
lib/lib_basic/NeoPixelBus-2.6.3/src/internal/Esp32_i2s.c:41:10: fatal error: soc/dport_reg.h: No such file or 
directory
 #include "soc/dport_reg.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

s-hadinger, 2021-06-08:
`dport_reg.h` still exists in C3 but changed location. I have read somewhere that esp-idf is changing these location progressively

Makuna, 2021-06-08:
Ok, another update to branch.  Found a similar solution to what I did within the esp32 includes.

Jason2866, 2021-06-08:
This does not solve. I did for testing just comment the line `#include "soc/dport_reg.h`
After doing that the `soc/sens_reg.h` and `driver/dac.h` are not found. After commenting this
too, i had the errors i uploaded above.

Makuna, 2021-06-08:
ok, missed that.  They have seemed to completely reinvent i2S interface and structures.  So I just removed the support for i2s completely for ESP32C3.

Jason2866, 2021-06-09:
With this changes https://github.com/Makuna/NeoPixelBus/pull/478 it does compile

Makuna, 2021-06-09:
@Jason2866 It should work as coded unless there is something incorrect with your build.  CONFIG_IDF_TARGET_ESP32C3 should only be defined for C3, why is it defined in your build system?

Jason2866, 2021-06-09:
We build for the C3 so `CONFIG_IDF_TARGET_ESP32C3` is defined. See
https://github.com/arendst/Tasmota/blob/development/lib/libesp32/LITTLEFS/src/esp_littlefs.c#L66-L76
which works as expected

Makuna, 2021-06-09:
Jason, I will not merge those in.  They don't follow the pattern I use and the changes should not be needed.  

The issue is that that mess at the top of the file esp_littlefs.c for the IDF Version include is setting the define for that library and it is not being defined otherwise.

The IDF uses the file sdkconfig.h to define these.  This approach (of using a header file) means this file must be included in EVERY c/cpp file; this is the primary reason Arduino.h is included to catch these sorts of solutions.  With Arduino IDE builds this should get handled for the user, for PIO, this had to handled by the project and probably should be included in its Arduino.h file or some solution that ends up including it (like esp_idf_version.h)

#include "sdkconfig.h" // this sets useful config symbols, like CONFIG_IDF_TARGET_ESP32C3

Jason2866, 2021-06-09:
Your approach is better! By doing this you avoid possible problems in future.

s-hadinger, 2021-06-09:
Thanks @Makuna, it makes sense. Actually we can force `#include "sdkconfig.h"` at project level for each c/cpp file. I have done this already for int/long int issue.

s-hadinger, 2021-06-09:
One last change needed, `NeoEspBitBangMethod.cpp` needs to be disabled for Esp32C3 too. There is a compilation error I wouldn't want to take time in solving it.

I'm now forcing `#include "sdkconfig.h"` anyways.

Makuna, 2021-06-09:
probably something to do with cycle count (suspect the opcode is not supported due to core chip changes).

Done.

s-hadinger, 2021-06-09:
This is good for me, all working now. I will update Tasmota once this is merged.

Thanks again for responding so quickly!

closed 2021-06-09 by s-hadinger