Makuna/NeoPixelBus #485: breaking change | esp8266 sdk version >= 3.0.1 | I2S.h

coelner, 2021-06-14:
With the esp8266 sdk version 3.0.1 onwards: https://github.com/esp8266/Arduino/pull/8108
will break this line: 

https://github.com/Makuna/NeoPixelBus/blob/2ff885c702a6d9f7a7ce6a80fd48512fa7c71085/src/internal/NeoEsp8266DmaMethod.h#L42

Makuna, 2021-06-14:
https://github.com/Makuna/NeoPixelBus/pull/486

akarnani, 2021-07-01:
Would it be possible to cut a release with the above pull request included?  It would be nice to be able to pull in these changes with the library manager.

coelner, 2021-07-01:
The release 3.0.1 is officially available: https://github.com/esp8266/Arduino/releases/tag/3.0.1

I do not know any method to determine which version is installed and select through a precomiler flag the right code snippet.

Makuna, 2021-07-01:
v2.6.5

closed 2021-07-01 by Makuna