Makuna/NeoPixelBus #492: RMT Channel for ESP32-C3 should be 0-1 and not 3

obaterspok, 2021-07-02:
**Describe the bug**
NeoEsp32RmtMethod.h file has the following comment:

// RMT channel 3 method is the default method for Esp32S2 & Esp32C3

I believe the comment is only valid for S2 which has 4 RMT channels for transmitt/Receive while the ESP32C3 also has 4 channels only the first two (0-1) can be used for transmit while channel 2-3 can only be used for receiving.

_"The RMT has four channels numbered from zero to three. The first half (i.e. Channel 0 ~ 1) channels can only be configured for transmitting, and the other half (i.e. Channel 2 ~ 3) channels can only be configured for receiving. They are referred to using indexes defined in structure rmt_channel_t."_

**Development environment (please complete the following information):**
- Latest main

Makuna, 2021-07-05:
https://github.com/Makuna/NeoPixelBus/pull/495

Makuna, 2021-07-07:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.6.6

closed 2021-07-07 by Makuna